prusa3d/PrusaSlicer #4261: PS crashes when clicking on Search in Print-, Filament- and Printer-Settings Tab

foreachthing, 2020-05-18:
### Version
2.3.0-a0
Commit: #[5c1d736](https://github.com/prusa3d/PrusaSlicer/commit/5c1d73682d8e347a7dd06c9ae8b1025f3e7a9021)

### Operating system type + version
<details>
  <summary>System Info</summary>
PrusaSlicer
Version:   2.3.0-alpha0
Build:     PrusaSlicer-2.3.0-alpha0+UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition
Total RAM size [MB]: 34,019MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.66
Vendor:       NVIDIA Corporation
Renderer:     Quadro P2000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
[... 327 lines cut ...]
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### Behavior
* Using the Search-button (or shortcut CTRL+F) while in 3D-Editor-View, Search works as expected.
  * Although, I would not expect _**pattern**_ to find _Com**pat**ible prin**ter**s co**n**dition_.
* Using Search or shortcut while in another settings tab, PS crashes.

No output from console with loglevel 0-5.

YuSanka, 2020-05-18:
Thanks for your report. This bug is fixed in https://github.com/prusa3d/PrusaSlicer/commit/5c142ccfd6b575803e65d577f73bb47994c8df2a

closed 2020-05-18 by YuSanka

foreachthing, 2020-05-19:
Thanks @YuSanka! Works like a charm!