prusa3d/PrusaSlicer #4263: Changing wall printing direction every layer in Prusa Slicer?

jsourice, 2020-05-18:
Hi everyone,

I have been sent here from Prusa Slicer forum, maybe someone can help me ! I'm new to github too, and while I usualy try to make things right, I'm maybe in the wrong section to post this demand as it's not actualy an "issue".

I'll just copy a part of my message from the forum to fully explain what problem I encounter. If you want the full context, here is the original post : [Changing wall printing direction every layer in Prusa Slicer?  
](url)

Best,

> Maybe someone can help us, here is one scenario :
> 
> I'm printing a square base tower, 3mm sided, 100% infill. Our Zetamix printing condition are almost the same as any type of regular plastic filament, except printing temp is much lower and overlap from the infill on the walls is fat as f***. 
> Now the problem : **I want the wall to be printed alternatively in clockwise and anti clockwise direction every layer.**
> Do you have any idea how to do that, natively or not, on Prusa slicer ? We already have a backup plan by manipulating the gcode itself, but as you know it would be a freacking pain.

Julien

bubnikv, 2020-05-19:
> I want the wall to be printed alternatively in clockwise and anti clockwise direction every layer.

Why? Doing that may produce visible artifacts on the outside wall.

jsourice, 2020-05-19:
> Why? Doing that may produce visible artifacts on the outside wall.

Yes, it may needs a quick explanation ! We are printing with high content of ceramic (more than 80%w) and then we debind and sinter the parts. We observe that when we print such small pillars, they tend to twist while sintering, always in the same rotation direction that the print. So we were looking for a rapid solution to counter that effect by switching clock / anticlockwise printing direction layer, hopping that this can greatly reduce this undesirable effect.

I understand it may affect the overall surface quality but in our case it's not the main concern for these type of parts at the moment, as the twist is much more detrimental to the part.

Thanks !

SimonSolar2C, 2021-06-04:
Hey Julien @jsourice , did you ever find a solution - SuperSlicer may have one with its outer wall vase mode.
Im about to try it on a big print because the direction changes at various positions in the walls are leaving artifacts

jsourice, 2021-06-04:
> Hey Julien @jsourice , did you ever find a solution - SuperSlicer may have one with its outer wall vase mode.
> Im about to try it on a big print because the direction changes at various positions in the walls are leaving artifacts

Hey,

Yeah we find a solution with our partner Raise3D, they added this change in their last update few weeks after we ask them, very very reactive team there ! So you may want to try IdeaMaker and test this feature, prints are amazing now. Our Zetamix filaments are often loaded with 80%w or + of ceramics and/or metal, so the effect of printing always in the same direction was strong enough to heavily twist the part after sintering, now it's all gone !

SimonSolar2C, 2021-06-04:
Many thanks for your prompt reply - I'll take a look at Ideamaker

webs02, 2024-12-28:
With this setting in Cura, each perimeter in the layer is printed in an alternating direction. And in each layer above, the perimeter above is printed in the opposite direction. 

This is an extremely helpful function for many people to be able to produce distortion-free prints. Especially when ASA or ABS is to be printed. 

Please check again whether this function can also be integrated into PrusaSlicer.

closed 2026-09-29 by prusaslicer-bot[bot]