prusa3d/PrusaSlicer #4264: 2.2.0 does not build with boost-1.73

kobboi, 2020-05-18:
### Version
PrusaSlicer 2.2.0

### Operating system type + version
Gentoo Linux amd64 unstable 20200518
Boost 1.73

### Behavior
[ 16%] Building CXX object src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o
cd /var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-2.2.0_build/src/admesh && /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_LOG_DYN_LINK -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-version_2.2.0/src -I/var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-2.2.0_build/src/platform -I/var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-version_2.2.0/src/clipper -I/var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-version_2.2.0/src/polypartition -isystem /usr/include/eigen3   -O2 -march=x86-64 -pipe -fomit-frame-pointer -g -ggdb -fext-numeric-literals -Wall -Wno-reorder -fPIC   -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -std=gnu++17 -o CMakeFiles/admesh.dir/stlinit.cpp.o -c /var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-version_2.2.0/src/admesh/stlinit.cpp
/var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-version_2.2.0/src/admesh/stlinit.cpp:31:10: fatal error: boost/detail/endian.hpp: No such file or directory
   31 | #include <boost/detail/endian.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/admesh/CMakeFiles/admesh.dir/build.make:135: src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o] Error 1
make[2]: Leaving directory '/var/tmp/portage/media-gfx/PrusaSlicer-2.2.0/work/PrusaSlicer-2.2.0_build'
make[1]: *** [CMakeFiles/Makefile2:804: src/admesh/CMakeFiles/admesh.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

kobboi, 2020-05-18:
Was compiling fine with boost-1.72.0

kobboi, 2020-05-18:
Might need something similar as done in https://github.com/PointCloudLibrary/pcl/commit/dbadf4143bdc503203da0d87a786752a60d29e76

adrien-n, 2020-05-29:
Your commit fixed the issue for me, thanks.

hroncok, 2020-06-01:
I actually get a test failure on s390x (big endian) with https://github.com/kobboi/PrusaSlicer/commit/d982345f52b6afbcc30c532828e161b2c956acf4:

```
test 4
    Start 4: libslic3r_tests
4: Test command: /builddir/build/BUILD/PrusaSlicer-version_2.2.0/Build/tests/libslic3r/libslic3r_tests
4: Test timeout computed to be: 10000000
4: Testing sort_remove_duplicates
4: Passed in 1.2e-05 [seconds]
4: 
4: Testing string_printf
4: Empty format with empty data should return empty string
4: String output length should be the same as input
4: String format should be interpreted as with sprintf
4: String format should survive large input data
4: Passed in 7.8e-05 [seconds]
4: 
4: Testing Scenario: Reading 3mf file
4:     Given: umlauts in the path of the file
4:      When: 3mf model is read
4:      Then: load should succeed
4: Passed in 0.135267 [seconds]
4: 
4: Testing Scenario: Export+Import geometry to/from 3mf file cycle
4:     Given: world vertices coordinates before save
4: facet (201)/ first facet
4:   vertex 0/201  7.56391317E+34 -5.28419300E+06  1.98271266E-09
4:   vertex 1/201  NAN  5.72225045E-10  1.98271266E-09
4:   vertex 2/201 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4: facet (202)/ second facet
4:   vertex 0/202 -3.08217422E-24  4.61318062E+05  1.98271266E-09
4:   vertex 1/202 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4:   vertex 2/202  NAN  5.72225045E-10  1.98271266E-09
4: facet (201)/ first facet
4:   vertex 0/201  7.56391317E+34 -5.28419300E+06  1.98271266E-09
4:   vertex 1/201  NAN  5.72225045E-10  1.98271266E-09
4:   vertex 2/201 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4: facet (202)/ second facet
4:   vertex 0/202 -3.08217422E-24  4.61318062E+05  1.98271266E-09
4:   vertex 1/202 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4:   vertex 2/202  NAN  5.72225045E-10  1.98271266E-09
4: facet (202)/ first facet
4:   vertex 0/202 -3.08217422E-24  4.61318062E+05  1.98271266E-09
4:   vertex 1/202 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4:   vertex 2/202  NAN  5.72225045E-10  1.98271266E-09
4: facet (201)/ second facet
4:   vertex 0/201  7.56391317E+34 -5.28419300E+06  1.98271266E-09
4:   vertex 1/201  NAN  5.72225045E-10  1.98271266E-09
4:   vertex 2/201 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4: facet (202)/ first facet
4:   vertex 0/202 -3.08217422E-24  4.61318062E+05  1.98271266E-09
4:   vertex 1/202 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4:   vertex 2/202  NAN  5.72225045E-10  1.98271266E-09
4: facet (201)/ second facet
4:   vertex 0/201  7.56391317E+34 -5.28419300E+06  1.98271266E-09
4:   vertex 1/201  NAN  5.72225045E-10  1.98271266E-09
4:   vertex 2/201 -1.49561025E+19  2.33332894E+17  1.98271266E-09
4: facet (203)/ first facet
4:   vertex 0/203  NAN  5.72225045E-10 -9.32876945E-33
4:   vertex 1/203  3.85977361E-12 -2.20106877E-02 -9.32876945E-33
4:   vertex 2/203 -3.08217422E-24  4.61318062E+05  1.98271266E-09
4: facet (239)/ second facet
4:   vertex 0/239  3.85977361E-12 -2.20106877E-02 -9.32876945E-33
4:   vertex 1/239  NAN  5.72225045E-10 -9.32876945E-33
4:   vertex 2/239  7.56391317E+34 -5.28419300E+06 -9.32876945E-33
4: facet (203)/ first facet
4:   vertex 0/203  NAN  5.72225045E-10 -9.32876945E-33
4:   vertex 1/203  3.85977361E-12 -2.20106877E-02 -9.32876945E-33
4:   vertex 2/203 -3.08217422E-24  4.61318062E+05  1.98271266E-09
4: facet (239)/ second facet
[... 1908 lines cut ...]
4:     Given: in ASCII format
4:      When: line endings CRLF
4:      Then: load should succeed
4:     Given: in ASCII format
4:      When: nonstandard STL file (text after ending tags, invalid normals, for example infinities)
4:      Then: load should succeed
4: Testing CGAL and TriangleMesh conversions
4: Passed in 2.0202 [seconds]
4: 
4: Testing ISO8601Z
4: Passed in 0.000133 [seconds]
4: 
4: Testing Slic3r_UTC_Time_Format
4: Passed in 1.8e-05 [seconds]
4: 
4: Testing Negative 3D offset should produce smaller object.
4: Elapsed processing time: 1.50538
4: Passed in 1.51638 [seconds]
4: 
4: ===============================================================================
4: test cases:  31 |  29 passed | 2 failed
4: assertions: 307 | 305 passed | 2 failed
4: 
4/7 Test #4: libslic3r_tests ..................***Failed    4.32 sec
```

lukasmatena, 2023-02-09:
Closing. Solved long ago with 3a194f91, we are now using Boost 1.78.

closed 2023-02-09 by lukasmatena