prusa3d/PrusaSlicer #4266: Auto orientate on import to reduce overhang and improve print quality

JamesRHarris, 2020-05-18:
2.2.0+win64
Windows 10 pro 19628.1
Prusa mk3s

On stl import the model is often oriented on the build sheet poorly.  There should be a tool to orientate the model optimized to print. Simply the orientation that reduces the most amount of overhang and needs the least amount supports should be default. I would also suggest this option be a default in the simple profile as good orientation greatly improves build quality and success.

New Feature Request.

Animo78, 2020-05-26:
It would be awesome if there was an Auto-Rotate button, so you can use minimal support.

gcarvalho, 2021-01-20:
This works reasonably well in it's CURA-specifc implementation : https://github.com/ChristophSchranz/Tweaker-3

nyxaria, 2021-09-26:
This would be so useful, especially since this and tree supports are the only things stopping PrusaSlicer from being the best slicer out there imo

saucesome, 2023-02-09:
Does this make more sense now with organic supports?

dmsheets377, 2023-06-29:
Support this for sure, makes even more sense now that we also got organic support for FDM. No reason this math should only be available for SLA

subworx, 2023-11-06:
I would, in addition, request to have an auto-orient function that optimizes for print time.
E.g. I have an STL (yet another fidget infinity cube v2) which prints 2:50h when imported and orientated horizontally along the print sheet (or after using auto align), and only 2:45h if I rotate it by 45 degress into any direction.
Having this done automatically instead of reslicing the object at various angles to find the shortest time would be awesome.

nyxaria, 2026-09-02:
still relevant