prusa3d/PrusaSlicer #4267: Wall bevel issue

gbzh01, 2020-05-19:
### Version
2.2.0+

### Operating system type + version
Mac OS 10.15.4

### 3D printer brand / version + firmware version (if known)
Prusa MK3s MMU2s, latest drivers

### Behavior
* _Describe the problem_
Bevel on a wall objects gets unpredictable results. Always the same bevel on that object -> must be something with the path planning.

* _Steps needed to reproduce the problem_
I can slice it with fixed layer height or variable - it never will be good

* _Expected Results_
Even bevel on that corner as on the other 3. The printer can do it nicely on the other 3 so it must be possible on that one too

* _Actual Results_
Unpredictable result but never acceptable.

 
[image]
[image]

neophyl, 2020-05-19:
Have you considered it may be a cooling issue ?  With the Mk3 fan only being on the front it is possible depending on the model design and orientation to get less effective cooling in some areas.

If you rotate the model does the problem also move with the rotation ?

gbzh01, 2020-05-19:
I rotated the model and I had no success. The effect happened by rotating as well. I could increase the cooling. That may be a point to check.
These are my actual settings for cooling (standard):

[image]

gbzh01, 2020-05-19:
I am running a test now with increased cooling and will report back

rtyr, 2020-05-19:
Please provide 3MF.

gbzh01, 2020-05-19:
[image]

Solved! It was the cooling. I increased from 30-50% to 50-80%. I could hug you neophyl! I really had a hard time with that! Thank you so much.

closed 2020-05-19 by gbzh01