prusa3d/PrusaSlicer #4271: Please support Slicing and Export for Anycubic Photon S

ngroulx, 2020-05-19:
### Version
_Version of PrusaSlicer used goes here_
**2.2.0**
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
**Windows 10 (build 18363), 64-bit edition**

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
**Anycubic Photon S**
**Firmware v5.0.2**
### Behavior
* _Describe the problem_
Can't directly slice and export into a format supported by Photon S. Instead need to export into an .STL (including supports) then load the .STL into photon's own software in order to finish the slicing and export into their .pws format. Would like to be able to do this all directly from within Prusa Slicer! ie: Please support AnyCubic Photon S in Prusa Slicer. 
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
**YES**

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-05-19:
This is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3713. We are leaving this task to contributors.

closed 2020-05-19 by rtyr

czo, 2023-02-09:
I recently got a Photon S, and their 'pws' format is seems very-very similar to the already supported 'pwmx' format. I would like to implement it. There are anyone working on it?