prusa3d/PrusaSlicer #4275: Disable / control bottom interface layers

Pun-e, 2020-05-20:
### Version
2.2.0 Win64

### Behavior
When generating support, interface layers make a dramatic difference to the quality of the supported layer **but** in my experience, the bottom interface layers almost always cause more problems than they prevent.
I'm printing links in a chain, and the holes have chamfers to facilitate insertion of the link pins. You can see that interface layers are being generated all along the chamfer as is "intended":
[image]
This is what the finished interface looks like:
[image]
The problems begin when the print is finished. These bottom layers are **completely** fused with the part, and cannot be removed. The following parts were printed in PETG with 0.3mm layer heights:
[image]
After "removal" (left side is the top, and right side is the bottom):
[image]
This picture was taken after hand removal, then pliers, _then_ sidecutters, **then** using a special tool to drive out the rest of the "support" through the hole. You can see that the surface has now shattered, and the part is completely unusable.
The default functionality has completely destroyed every print I've ever had that has any bottom interfacing layers, I struggle to even understand why it works the way it does. Turning off interface layers allows for easy removal, but the surface finish is sub-par. This cannot be worked around using height-range modifiers either.
Thanks for checking this out, and hopefully this is something you'll be able to take care of

liftbag, 2020-05-20:
Slic3r ++ has separate settings for top and bottom supports separation. It would not be bad to include this feature in PrusaSlicer as well.

jon-le0234, 2021-02-17:
I agree, removing interface layers is a pain on PETG, especially for my one design which requires supports in a cavernous area with a gentle slope. -_-"

@liftbag Where would those settings be in Slic3r? I checked but couldn't find it. Cura and Simplify3D also don't have separate controls for contact distance between the interface layer and the object.

I could settle for what Cura has, which is the ability to toggle the roof or floor for the interface layer.
[image]

liftbag, 2021-02-20:
> @liftbag Where would those settings be in Slic3r? I checked but couldn't find it. Cura and Simplify3D also don't have separate controls for contact distance between the interface layer and the [...]

[image]

bubnikv, 2021-03-16:
73b88e6ce0fa21a031490f148751114af73d573c

    Splitting FDM support gap to top / bottom, introducing
    support_material_bottom_contact_distance

[image]

There are still some glitches in the master, but the chances are good.

jon-le0234, 2021-03-17:
I was able to generate the g-code without the bottom interface layers. Hooray! Thanks, @bubnikv!

However, I'm encountering a second obstacle. There is still another dense interface layer at the bottom, which is generated at the base of all supports. Is there a way to control that one as well?

Full model with supports:
[image]

Base support layer:
[image]

Support settings:
[image]

bubnikv, 2021-03-22:
That should be fixed with 6f3f3624c54e3cf2b508ca4558f2357604318d81

bubnikv, 2021-03-22:
And for my reference, this issue is a duplicate of #4288

closed 2021-03-22 by bubnikv