prusa3d/PrusaSlicer #4276: printing order

Voodootec, 2020-05-20:
### Version
2.0.0+win64

I don't know if it's really a problem but I noticed something. If I copy the same object multiple times and then align them, everything is normal. But the copies are not aligned (see image).  Normally this shouldn't be a problem but it disrupts the printing order and in the worst case, it delays the printing time. In my example, i expect it to start on one side but he prints across the bed. And that extends the printing time. And it changes the prioritization in every layer.

As an example, he start at 1 going to 5, than to 2, than 3, than 6 and at the end to 4. 

Maybe someone has already observed this?

[image]

JohnnyDeer, 2020-05-20:
Hi @Voodootec , this issue is duplicate of #2474 so i´m closing this one.

closed 2020-05-20 by JohnnyDeer

Voodootec, 2020-05-20:
oh shit, i'm sorry. Thank you for your replay.

JohnnyDeer, 2020-05-20:
@Voodootec no worries :)