prusa3d/PrusaSlicer #4277: Custom mouse bindings

ryucz, 2020-05-20:
### Version
2.2.0

Featrure request:

I would like to be able to set my custom mouse buttons bindings to match my other 3D software.

Pan/Zoom/Orbit should be cutomizable.

JohanAR, 2021-01-01:
I just had a look at the function where this is handled and it's 400 lines of nested if-else clauses with hard-coded mouse buttons, so I think it would take quite some effort to implement this.

I could make a try if I knew the devs didn't mind a major refactoring of GLCanvas3D::on_mouse

slipjigpoppys, 2021-03-20:
I may be off base here, but in looking through that function it looks as if the panning and orbiting is done in the following else if chain: 
(slic3r\GUI\GLCanvas3D.cpp Ln 3344)
```c++
        else if (evt.LeftIsDown()) {
            // if dragging over blank area with left button, rotate
            [...]
        }
        else if (evt.MiddleIsDown() || evt.RightIsDown()) {
            // If dragging over blank area with right button, pan.
            [...]
        }

```

Which, if that's the case and there's no Gotcha (maybe a big If), then perhaps something like this would work:

```c++

//Define some styles of pan/orbit, presumably from other programs
typedef enum mouse_pan_orbit_setting_e 
{
    pan_orbit_default,
    pan_orbit_cura
    pan_orbit_fusion360,        
} mouse_pan_orbit_setting_e;

mouse_pan_orbit_setting_e mouse_pan_orbit_type = pan_orbit_default; 

// get a setting based on, presumably based on something in the preference menus
mouse_pan_orbit_type = mouse_pan_orbit_preference_setting;

//Some placeholders for what the else ifs are checking for.
//(Not sure if evt.LeftIsDown() etc actually returns a bool)
bool mouse_orbit_is_down    = false;
bool mouse_pan_is_down      = false;

//Change up the trigger flags based on desired settings
switch(mouse_pan_orbit_type)
{       
    case pan_orbit_default:
        mouse_orbit_is_down = evt.LeftDown();
        mouse_pan_is_down   = (evt.RightDown() || evt.MiddleDown());
    break;

    case pan_orbit_cura:
        mouse_orbit_is_down = evt.RightDown();
        mouse_pan_is_down   = evt.MiddleDown();
    break;

    case pan_orbit_cura:
        //this one may not work b/c of the Shift+Click that might be handled elsewhere
        mouse_orbit_is_down = evt.MiddleDown() && wxGetKeyState(WXK_SHIFT);
        if(!mouse_orbit_is_down) { mouse_pan_is_down   = evt.MiddleDown(); }        
    break;

    default:
    break;
}

```

with the relevant else if flags replaced like so:

```c++
        else if (mouse_orbit_is_down) {
            // if dragging over blank area with left button, rotate
            [...]
        }
        else if (mouse_pan_is_down) {
            // If dragging over blank area with right button, pan.
            [...]
        }

```

Might be worth a compile and check... (which i am conveniently not setup to do...)

JohanAR, 2021-03-20:
@slipjigpoppys I was thinking some similar logic, but probably broken out into a separate control scheme class so that it would be easier to further extend it. There are some things that complicate it a little, for example gizmos and mouse buttons having different functions depending on if the mouse is over the plate or an object, but I think it should definitely be doable. Been quite busy with other projects, but I actually just started looking at this again yesterday evening, though as a patch for SuperSlicer after having had a chat with Merill. If you're interested in joining the effort, come visit the SuperSlicer discord

towlerg, 2021-07-03:
No where near as clever as the guys above but perhaps a usage case might prompt, I use a CAD package (DesginSpark which is a free version of a commercial product, the dev's are unavailable) that does not have configurable Pan and Rotate. It would be great if both slicer and CAD package used the same keys!

JohanAR, 2023-01-11:
I did not continue working on this. Haven't checked the code if any Prusa3D devs have done any changes to the input handling that would make customisable mouse buttons easier to implement, or if it still looks the same (or even worse :))

DrunkenJediMaster, 2023-01-11:
+1 to this request

IMO, all software that manipulates 3D objects should have this customizable option.

strepto42, 2023-07-12:
+1 why is this still even a question

BoJl4apa, 2023-07-17:
+1 to this request

MKloberg, 2023-08-06:
+ 1 here too. This would be a super nice to have in the preferences.

foreachthing, 2023-08-06:
> I design a lot and coming from CAD into the slicer always throws me off a little.

That's why I got myself a 3Dx SpacePilot. :-)
I don't understand my every developer has to invent new controls...

S13RA, 2023-11-01:
Yes please. Prusa slicer seems to be doing everything reversed from what i would expect. Why is rotate on LMB, when it is also used to manipulate objects? Why are MMB and RMB doing the same thing (panning)? 
Also the speed of orbit is super fast, while panning is very slow. Would be nice to have those 2 speeds more balanced (or also adjustable).
In short, a program like this should have customizable mouse controls.

(dont get me started on the lack of mouse-zoom-to-cursor, the weird center point of rotation when orbiting, or the fact that orbit also does a camera roll function)

codefoster, 2024-01-10:
> I just had a look at the function where this is handled and it's 400 lines of nested if-else clauses with hard-coded mouse buttons, so I think it would take quite some effort to implement this.
> 
> I could make a try if I knew the devs didn't mind a major refactoring of GLCanvas3D::on_mouse

That doesn't sound to me like it's necessarily difficult. The hard coded references should be refactoring anyway.

arslaman, 2024-01-22:
Is there any progress on this? I'm looking forward for this feature

kalvaer, 2024-03-09:
May 2020 -> March 2024, and this is still open?

There is no need to beat a dead horse here, but.... As so many have mentioned if you use almost any CAD software, its a complete pain to have to remember to flip your workflow around each time you switch between Prusa Slicer and CAD  windows.

aaroneiche, 2024-03-18:
I'll throw my hat in here and say that I'd really like this. I use FreeCAD for design work, which shares its default mouse controls with Blender. In that state the middle-mouse button is Orbit, and L+R buttons are Pan. So it's inverted.

crgriseus, 2024-04-05:
I just changed the navigation settings to "Blender based" for everything except Prusaslicer. I hope you'll update this soon.

it-guru, 2024-04-21:
+1 to this request

GADify, 2024-05-16:
Add me to the +1

Directory, 2024-06-09:
waiting on this as well

MichiMess, 2024-07-25:
Me 2! All the other 3d programs are working with right mouse button. Please fix this soon!

KSEGIT, 2025-01-30:
+1 This should be done!

SophisticatedLady, 2025-01-31:
+1 Revit, Rhino, & Slicer can't all be different

strepto42, 2025-02-02:
I've made a fork and am fixing this. Already implemented a simple left/right mouse button swap. Took literally half an hour (once I figured out how to build the code).

I'll post here and make a PR once it's a more complete solution and is tidied up.

strepto42, 2025-02-04:
Basic PR in. Thought I may as well as I'm not sure when I'll be able to get to it more thoroughly. https://github.com/prusa3d/PrusaSlicer/pull/14090

RYUcze, 2025-02-05:
And it only took 5 years :D

I hope the PR will be merged.

KSEGIT, 2025-02-05:
Nice! Were you able to build and test it?

movatica, 2025-02-24:
Thanx for working on this!

FreeCAD has a lot of different navigation styles to select from, you could borrow from there!
Find their source code right here: https://github.com/FreeCAD/FreeCAD/tree/main/src/Gui/Navigation

strepto42, 2025-08-25:
I accept payment in crypto 😅

olsoma13, 2025-08-25:
Are Prusameters a fungible currency yet?

strepto42, 2025-08-25:
Hah 0.0005 PsaXLcoin

One of these days I'd love to have time to refactor this properly a la movatica's comment above, or similar. Unfortunately they haven't merged my original PR yet but perhaps I'll try to update my fork and rebuild/release a build so at least others can use the basic mod.

strepto42, 2026-05-06:
I have a Snapmaker U1 now. I am happier. Although their custom slicer (forked from Orca) is a bit stinky in places. It *does* let you reverse the mouse buttons though... ¯\_(ツ)_/¯

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena