prusa3d/PrusaSlicer #4278: Wrong slicing of object with hole

Tomassino, 2020-05-20:
### Version
Version 2.2.0+linux-x64

### Operating system type + version
Arch linux, latest updates

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S MMU2S

### Behavior
I have a problem slicing an object I created. Depending on the "resolution" (i.e. the value of the `$fn` OpenSCAD variable) an hole in the model is created, not created at all or not created but its top perimeter is (more or less) identified. I have attached the source OpenSCAD file to reproduce the problem, the various stl files and screenshots of PrusaSlicer (3D and preview). I am using the default print settings for my printer (single spool configuration)

#### $fn = 25 - expected behaviour
[image]
[image]

#### $fn = 50 - hole missing
[image]
[image]

#### $fn = 100 - hole top perimeter added (plus something else)
[image]
[image]

#### $fn = 200 - hole missing
[image]
[image]

#### Zip file with all images plus OpenSCAD source code and stl files
[bug slicer prusa 20200520.zip](https://github.com/prusa3d/PrusaSlicer/files/4658512/bug.slicer.prusa.20200520.zip)

BryanSmithDev, 2020-05-20:
There is an issue with the models. After I ran them through NetFabb to repair them they sliced correctly.

Tomassino, 2020-05-20:
Oh, thank you very much for your help, it's the first time I have a problem like this with models created in OpenSCAD. I successfully fixed the model with blender, so I'm closing this issue as it is definitely not a bug in PrusaSlicer. Sorry for the wrong report.

closed 2020-05-20 by Tomassino