prusa3d/PrusaSlicer #4285: Cut hold instead of flat surface for part connection?

kenthinson, 2020-05-22:
### Version
2.2.0+

### Operating system type + version
Mac OSX 10.15.4

### 3D printer brand / version + firmware version (if known)
Ender 3, Klipper firmware

### Behavior
Hello. When I print and there is a small part on top of a flat surface I get solid infill, then the part is printed on top. The problem I'm having is that it doesn't stick very well and comes off. Im' curios if I can make the slicer cut some kind of hole so the infill into the upper part is one continuous piece. 
Sorry if this is a obvious thing. I am not even sure what I should be googling for this.

[image]
[image]
[image]

All projects using default settings.

neophyl, 2020-05-22:
Add a modifier inside that area and increase either the perimeters or infill % (or both).  However if that hinge part of the Sliding Legolini is breaking off then I would suggest that you should possibly increase your print temperature 5-10 degrees.  
None of mine have broken :)

kenthinson, 2020-05-22:
@neophyl your printed yours with Prusia Slicer? I have not tried this part yet but have experienced breakage on similar parts when they are attached to the flat top. Cura seems to do what I am talking about automatically. hmmm

bubnikv, 2021-06-03:
Isn't it a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/89 ?

kenthinson, 2021-06-03:
@bubnikv It appears so.

bubnikv, 2021-06-04:
@kenthinson thanks. Let's close this one then. It will become referenced from the other issue.

closed 2021-06-04 by bubnikv