prusa3d/PrusaSlicer #4288: Separate counts for floor and roof support interface layers

fpikus, 2020-05-22:
### Version
_Version of PrusaSlicer used goes here_
2.2.0

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5

### Behavior
Dense interface layers are often required to support parts that rest on supports and don't have much area on the initial layer (like elbows pointing down). But then I get the same number of interface layers on the bottom of the supports, between the model and the supports, and those are much harder to remove. There should be a way to not have any interface layers below supports, at least, if not a separate count. Also, a larger gap option would be nice. I care about how good the first layer of the model above the support is. The first layer of supports above the model can be rough, as long as it sticks and the supports build up on it, I don't care what it looks like.

_Is this a new feature request?_
Yes.

rtyr, 2020-05-31:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3772.

bubnikv, 2021-03-19:
fixed with 6f3f362

closed 2021-03-19 by bubnikv