prusa3d/PrusaSlicer #4290: [Feature Request] Small infill

qwewer0, 2020-05-23:
### Description

It would be useful if the slicer would have the following settings:
1. Small infill size
2. Small infill speed

### Small infill size
This would be found in the **Infill** section under **Advanced**, to modify what infil movements smaller than the set value would be printed with the "Small infill" speed.
It would reduce the printers shaking/vibration when printing the small infills "too" fast.

### Small infill speed
This would be found in the **Speed** section under **Speed for printing moves**, to modify the speed of the small infill printing movements.
Could be set as mm/s or % from the **Infill** speed and wouldn't/would (?) be overwritten by **Top solid infill** speed.

If this would cause the gcode to be a mess, then we can forget this.

BeatSlayer, 2020-05-23:
Isn't that what gap fill already does?

qwewer0, 2020-05-23:
> Isn't that what gap fill already does?

@BeatSlayer You are absolutely right, but only with the "Small infill speed" part, as I can't seem to find an option to change the "small gap" size.

So we are halfway there 😄

BeatSlayer, 2020-05-23:
That's "min gap radius" or something close

qwewer0, 2020-05-23:
> That's "min gap radius" or something close

Can't find anything like that.

BeatSlayer, 2020-05-23:
The name may be a little off as I'm not by the computer but it's at the bottom of the advanced tab at the print settings.

qwewer0, 2020-05-23:
Under the Advanced tab I only have these:

- Extrusion width... (none of these are it)
- Infill/perimeter overlap
- Bridge flow ratio
- Slice gap closing radius (cracks smaller than 2x gap closing radius are being filled during the triangle mesh slicing. ... this isn't either)
- Resolution
- XY size compensation
- Elephant foot compensation
- Clip multi-part objects

None of those are for that gap size option.

BeatSlayer, 2020-05-23:
"Slice gap closing radius"

That's the one I was talking about, and it's the one controlled by the Gap Fill speed

qwewer0, 2020-05-23:
That is for mesh gaps and not for wall to wall gap fill, if I'm not wrong.
So, if it is at 0.05 than any gap in the model that is smaller than 0.1 will be smoothed out.

LeavOA, 2020-08-29:
[image]

Not exactly what you wanted but a start...

qwewer0, 2020-08-29:
@LeavOA That isn't what I want, and not even helps.

Closing this, as Marlin has a good alternative in the firmware.

closed 2020-08-29 by qwewer0

LeavOA, 2020-08-29:
@qwewer0 Sorry, Misunderstood your issue.