prusa3d/PrusaSlicer #4292: With large overhangs and supports, overhang is printed in pessimal directions

richo, 2020-05-24:
### Version
2.2.0+-202003211132

### Operating system type + version
MacOS 10.15.4

### 3D printer brand / version + firmware version (if known)
It's a Prusa MK3s but the issue I believe is in the slicer

### Behavior
I created my object in Fusion360. It is some interlocking parts so it has a finger and a slot. I enabled supports for both sides.

On the side with the slot, it creates supports as expected, but it does not lay the first bridging layer in an ideal direction, and earlier prototype which didn't go very well, but on this part it laid the bridging layer longways, which amounts to a bridge the entire width of the print bed, which definitely didn't go very well.

I tried to find a way to specify which way the bridge should be laid, but it seems that especially with supports, the slicer itself should be attempting to "anchor" one side in the support.

I'm very happy to work on this issue if someone can offer some pointers on where to start looking in the source!

thanks

#### Project File (.3MF) where problem occurs
This is the 3mf file
https://gist.github.com/richo/a23fd2e4d4e4bfcddcf7f491e37bacb6
[image]

JohnnyDeer, 2020-05-25:
Hi @richo , please upload 3mf file.

richo, 2020-05-25:
Hi @JohnnyDeer it's in that gist I linked, but I just noticed that you prefer it as a zip attached to the issue so I'll do that now.
[issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4678453/issue.3mf.zip)

richo, 2020-05-25:
Oh, one quick update. I found the Bridging Angle parameter in print settings, but it's global so it appears that with bridges at right angles there is no setting you can use that isn't a compromise. My intuition is that the slicer should be optimising for either the shortest path on the bridge, or maybe shortest path with a small angle to let it "build" out from a corner. but, as I said I am very new to this so I'm not trying to be very prescriptive.

pedjas, 2020-06-19:
Once I got into similar issue. I had object with top cover along narrow gap. Slicer tried to do bridges on long paths instead short path across the gap. It failed, of course.

closed 2026-09-29 by prusaslicer-bot[bot]