prusa3d/PrusaSlicer #4293: [Feature Request]: Ability to save changes to the default profiles without creating a copy

fwdixon, 2020-05-24:
### Version
2.2.0

### Operating system type + version
Windows 10 x64 Pro

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3 

### Behavior
Feature Request:

Add the ability to save edits to the default profiles. It is frustrating to have to create duplicate profiles to save octoprint settings, or make changes to filament temps or print speeds on the default profiles because they are locked for editing.

I-like-stuff, 2020-05-25:
I would also like this to be implemented

pedjas, 2020-05-26:
Why not simply create your own custom default profiles and always use (and change) them?

I think it is goot to have default profiles that are safe from changing. That is good reference point for all other settings.

fwdixon, 2020-05-27:
That is what you have to do now. The problem is, you end up having to wade through twice as many profiles, often due to a single change like a temp going up 10 degrees or OctoPrint settings. 

You can always reload default profiles through the configuration wizard if you need to, so having them locked for editing seems a bit excessive.

neophyl, 2020-05-27:
As the profiles are based on a cascading inherited system now when Prusa update them they automatically get changed.  That's why they introduced the Vendor system recently to allow different Vendor spaces/profiles that wouldn't effect each other for all the built in ones.  If you could edit the base ones (which you can by editing the .ini files directly) the changes at the lowest level are also passed down to anything that inherits from them so one change can effect many profiles.  When they are updated anything you changed would be wiped out.

rogery555, 2020-12-21:
God yes please allow making simple changes without creating new profiles.  Like I don't want skirts on my default profiles but I don't want two default profiles in my list

foreachthing, 2020-12-21:
I'd like a "system", where a new profile (B) just stores the changes from the parent-profile (A). That way you could disable brim (in (B)) and still benefit from the updates on (A).

Kachidoki2807, 2020-12-21:
And yet, it is easy:
[image]

Create your custom inherited profile, and then go back to the Configuration Wizard and uncheck the system preset you don't want to see anymore.
The power of the system preset is that you can keep track of what is changing over the time, and even see thoses changes with the lock/unlock icon. That was useful to me during the beta, because without this system I would never have seen that the new infill anchoring parameters were changed.

GrubbyZebra, 2020-12-21:
Except that with filament profiles, if the profile has the same name as a
default, even if the default is not installed, it will not show up in the
list.

Kachidoki2807, 2020-12-21:
I admit there are some tricky corner cases, and even more if you want to use the same name. But it is a common problem, system names are often reserved and the user can't use them. It is the case in compilers or even in operating systems. It allows to avoid confusion between system level resources and user level resources.

As the system preset can and are constantly updated (not only the Prusa's ones), I don't think it is a good idea to replace them. Hiding them but keeping the reference is much more powerful and safe. It allows to inherit of the updates from the manufacturer, especially the new non-existing yet parameters, without loosing all the previous custom adjustments.

Maybe the UI can be a bit improved to finely select what is visible and what isn't, but I guess there is also a lot to do internally to be able to have non-unique named profiles.

closed 2026-09-29 by prusaslicer-bot[bot]