prusa3d/PrusaSlicer #4294: Weird Z-Hop Issue

Dan018, 2020-05-24:
### Version
 PrusaSlicer-2.1.1+win64-201912101512

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
TronXY X5SA 2019 version
SKR 1.3 with TMC2130 drivers
Marlin 2.0.5.3 (latest)

### Behavior

Printing  a 20mm calibration cube ( [https://www.thingiverse.com/thing:1278865](url) ) with Z hop enabled gives very strange results.

[image]

The cube on the left is sliced with z-hop set at 0.3mm. The cube on the right has the exact same settings only z-hop is disabled (i.e. set to 0mm).

The defective cube on the left measures 35mm in height so it is almost as if the z axis rises for the hop, but then doesn't go back to the layer height after leading to the larger Z dimension and the poor layers.

The only thing I changed is the z-hop height so that's obviously the issue. The right hand cube measures exactly 20mm in all directions and other prints are defect free and dimensionally accurate when printed without z-hop so I'm confident the printer is fine.

rtyr, 2020-05-24:
I don't think that this problem is caused by PrusaSlicer. Please provide your 3MF project file and gcode file so we can check it.

Dan018, 2020-05-24:
I've attached the files you asked for. The gcode looks ok with first layer height at line 53, layer change to 0.54 (0.24 first layer + 0.3 layer height ) with a z hop of 0.3 ( to 0.84 and back to 0.54 ) occurring around lines 245 to 252.

I'll post this over on the Marlin Github as well.

[Cal Cube Z Hop Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4674007/Cal.Cube.Z.Hop.Issue.zip)

rtyr, 2020-05-24:
The g-code is fine as I expected.

The issue is probably with your printer or most likely with your machine limit values. I can see, that you are using machine limits from i3 MK3. If you leave it like this, machine limits from PrusaSlicer will override your default firmware limits. My guess is that your printer fails to move print bed up with those incorrect machine limits (acc 1000mm/s Z axis for example). I would suggest you to either set machine limits properly (=according to your default firmware values) or to switch g-code flavor from "Marlin" to "Reprap/Sprinter", which will disable machine limits in PrusaSlicer. Then reslice the gcode, restart the printer and try again. Or just restart the printer and try attached gcode with commented M20X commands (see screenshot).

[image]
[xyzCalibration_cube_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/4674084/xyzCalibration_cube_fixed.zip)

Closing, output from PrusaSlicer is OK.

closed 2020-05-24 by rtyr