prusa3d/PrusaSlicer #4295: support problem

nerilorenzo, 2020-05-24:
### Version
_Version of PrusaSlicer used goes here_
2.2.0

### Operating system type + version
_What OS are you using, and state any version #s_ windows

### Behavior
* _Describe the problem_
the support produces a not printable part in correspondence of the first interference layer

[image]

[raccordo T 3.zip](https://github.com/prusa3d/PrusaSlicer/files/4674265/raccordo.T.3.zip)

bubnikv, 2021-04-08:
[image]

[image]

bubnikv, 2021-04-08:
Fixed with 8fd731f7a094854770e42d7f075e3946681ed3b1
We now have a smarter sparse support zig-zag infill generator.

closed 2021-04-08 by bubnikv