prusa3d/PrusaSlicer #4297: Ubuntu 18.04 - Internal error: basic_string::_M_construct null not valid

camisatx, 2020-05-25:
### Version
2.2.0

### Operating system type + version
Ubuntu 18.04.4
4.15.0-101-generic

### Behavior
**Problem**: When starting the program, no printbed is shown (just a white background). The toolbar shows `PrusaSlicer-2.2.0+UNKNOWN based on Slic3r`. I am able to load and slice a model, but no model (either 3d view or sliced view) shows anything).

Clicking `Help` -> `System Info` shows the message `Internal error: basic_string::_M_construct null not valid` message (see image). Clicking OK force closes the program.

The program worked when I first downloaded and ran it. However, upon closing and reopening it, this error appeared. Restarting the computer did not resolve anything.

[image]

lukasmatena, 2020-05-26:
The UNKNOWN in the title bar indicates that you're not running a build from our buildserver. Did you compile PrusaSlicer yourself? Does the problem also appear with our official build?

camisatx, 2020-05-26:
Thanks for that background! I thought I had PrusaSlicer installed from the AppImage. In any case, downloading a fresh AppImage copy and running directly from that works like a charm. I will stick with that.

I appreciate the help! Closing this issue.

closed 2020-05-26 by camisatx