prusa3d/PrusaSlicer #4298: Order of the printing object in the GCode generation

JonathanVaillancourt, 2020-05-25:
Hello,

I have an issue regarding the printing order of the object with the newest Prusa Slicer.
I'm currently using the version 2.0.0 which is generating the GCode according to the order the parts are put in the slicer. So part 1 will be print first, part 2 will be print second and so on... no matter what is the position on the bed. This features was important for me as I use color change (not with multi-material setup).

But when I change color, the first thing it prints leaves a kind of small drip (excess of material) and as I'm printing logo this is a problem as I have always one that were not good. Therefore I have put what I call purge block (tiny parts that I will scrap) as part 1, so that the first part it prints is this one.

Now with the newest version of Prusa Slicer it doesn't respect the order parts are insert in the slicer. This is a major concern for me as I'm cannot find a way to make sure the purge block will be printed first.

Could you please help improving the slicer as it was before or letting me know how I can manage which parts to be printed first.
Best regards.

lukasmatena, 2020-05-26:
Did you try using multilayer skirt to do the purging? It was originally designed to work as poor man's purge tower and should be always printed first.

If it has more loops and more tools are used on a layer, the skirt should in fact be the first thing that the new tool touches, but that may not be relevant for your setup. And there may be some issues with it too, I'm not sure how well it is tested and if anyone actually ever used it.

pedjas, 2020-05-26:
On the right side there is a list of objects. You can change their order by drag and drop. 
This does not change printing order?

JonathanVaillancourt, 2020-05-27:
@lukasmatena I think you are right that would work using multilayer skirt as I always see it be printed first, however I think it will create a lot of scrap material and I try to minimze the waste.
Unless I misunderstand you explanation, in that case let me know.

JonathanVaillancourt, 2020-05-27:
@pedjas with the new version of prusa slicer the list of objects on the right side doesn't matter. I also put my "purge block" first and it is not printing it first. I even increase the quantity of purge block to have 2 of them and sometime 1 is the second parts it prints and the second one is the last parts to print. In version 2.0.0 it follows exactly the order in the right side list.
I have the impression that the way the Gcode is generate that it try to minimize movement of printing head, but just guessing.

pedjas, 2020-07-23:
Now I stumbled to this issue. I have to pri9nt abut 50 copies of the seem small object. I add one STL to the plate and then multiply it and auto arrange. 

It prints parts in random looking order so substantial time is wasted moving head from one side to the other. It would be great if it prints them one next to another saving time. 

It seems we are unable to set order other than loading each copy as new STL and manually set it on the plate (did not try, just read that is the only way).

PrusaSlicer should have option to autoorder items so head spends minimal time moving from one part to another and option for user to manually set printing order (by rearranging list on the right for example)

closed 2026-09-29 by prusaslicer-bot[bot]