prusa3d/PrusaSlicer #4299: voronoi_ctypes.hpp(41,18): error C4235: nonstandard extension used: '__int128' keyword not supported

m3cinc, 2020-05-26:
### Version
2.3.0alpha0

### Operating system type + version
win10 Visual Studio 2019

### Behavior
Build fails due to multiple errors C4235 introduced by boost\polygon\detail/voronoi_ctypes...

* _Steps needed to reproduce the problem_

build according to Windows Noob recommendation the dependencies
then

mkdir build
cd build
cmake .. -G "Visual Studio 19" -DCMAKE_PREFIX_PATH="c:\PrusaSlicer\PrusaSlicer-master\usr\local"
open Visual Studio for c++ development (VS asks this the first time you start it)
Open->Project/Solution or File->Open->Project/Solution (depending on which dialog comes up first)
click on c:\PrusaSlicer\PrusaSlicer-master\build\PrusaSlicer.sln
Debug->Start Debugging or Debug->Start Without debugging PrusaSlicer reports multiple errors

bubnikv, 2020-05-26:
You are living on the edge. We did not update our build server yet.

Is it 32bit or 64bit build or both?

jmd73, 2020-05-26:
I have the same problem, @bubnikv. I tried building it yesterday again, but to no success.
The problem came after your commit 0dfa83f7dae8df732d5066d44e4962f4732a902c (in prusa3d/polygon) in the file  include/boost/polygon/detail/voronoi_ctypes.hpp. On row 41 is the offending define that the compiler does not like ("typedef unsigned __int128 uint128;").

I have no idea how to solve the problem. I don't usually code in visual studio.

The deps build without problem though.

I am compiling with Visual Studio 2019. I am doing a 64bit build.

I am at work now, so I can't send build logs or anything else. At most I can try to answer any questions you might have without the logs.

m3cinc, 2020-05-26:
Vojtech,
I was trying to build 64bit version and also implement a few changes for
mmu2 with skinnydip additions.

m3cinc, 2020-05-26:
I believe the problems can go away with a few modifications of the voronoi_ctypes.hpp file located in deps\build\destdir\usr\local\include\boost\polygon\detail directory 👍 
add the following in line14
#include <boost/multiprecision/cpp_int.hpp>
and modify line 41 to read
typedef boost::multiprecision::uint128_t uint128;

then, also fix line 519 to read
    ret_val = 0;

this will allow to complete the compilation without error.
I was able to build Release and Debug 64bit versions

The attached file extension of .txt was added to allow attachment of the modified file. just remove the extra extension and place voronoi_ctypes.hpp in the folder indicated above.
[voronoi_ctypes.hpp.txt](https://github.com/prusa3d/PrusaSlicer/files/4685619/voronoi_ctypes.hpp.txt)

bubnikv, 2020-05-27:
I suppose on Visual Studio we should use
https://docs.microsoft.com/en-us/cpp/intrinsics/mul128?view=vs-2019

That should produce more efficient code than boost::multiprecision

m3cinc, 2020-05-27:
Honestly believe that the multiprecision tool from boost is better to handle at the moment.

bubnikv, 2020-05-29:
> Honestly believe that the multiprecision tool from boost is better to handle at the moment.

Check the implementation. It uses 32bit limbs for multiplication on a 64bit system, which is quite a bad idea. Not everything boost is great.

bubnikv, 2020-05-29:
I took your advice and I added the boost multiprecision type. It is not the most efficient implementation, but if GMP is used, the branch using this int128 type is called rarely enough to not be important I hope.

This whole polygon excercise is to make the voronoi code faster and more stable, see
https://github.com/boostorg/polygon/issues

I am not sure yet whether all the changes I did are making things better or worse, the voronoi code is quite complex and it is too easy to introduce a bug which is difficult to track. I believe I have made the voronoi generator twice as fast on 64bit platforms than before using the GMP library.

closed 2020-05-29 by bubnikv