prusa3d/PrusaSlicer #4300: [Feature Request]: The ability to automatically insert 2 or so layers of higher infill before a top layer with large surface area

livingkurt, 2020-05-26:
### Version
2.2.0
### Operating system type + version
macOS Catalina 10.15.4

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s Firmware v3.9.0

### Feature Request
To save filament on prints that don't need to be as strong I like to use around 5% infill. But when using low infills (<10%), and printing top layers with large surface areas, If the print nozzle stops in between the infill lines it will begin to curl posing the risk of collision with the nozzle as well as leaving the top layer bumpy and unappealing. I propose the ability for Prusa slicer to automatically detect these issues and then give the option to the user to add 1 or 2 layers of higher percentage (<50%) infill right below each of the top layers where the issue seems likely. Doing this will create less probability that the nozzle will stop in-between the infill lines. 

Currently there is a way to do this manually by taking a slab modifier and shortening it down to about what 2 layers would be and then placing it right below the top layer that is concerning. Then give the modifier a higher infill value. It is not easy to accurately do this, for when making small adjustments the modifier slab jumps positions and sizes.

Using 5% Infill without a modifier slab
[image]

livingkurt, 2020-05-26:
Reuploading image Using 5% Infill without a modifier slab

[image]

milks808, 2020-12-09:
I'd also like to see this feature for the same reasons. I arrived at the same cumbersome workaround as livingkurt but a setting to quickly and automatically apply it would be great

ManuelMcLure, 2020-12-09:
I agree that curling filament on the first layer of non-sparse infill is the most common cause of failed prints/layer shifts for me, usually near the end of a long print. I think the suggested solution would go a long way to solving this issue.

togomedia, 2020-12-10:
Generally speaking, some form of variable infill, similar to "adaptive cubic" but more exponential or geometric in effect would solve this universally. Right now, I do not believe there is a way to adjust the rate (or steps) in the adaptive cubic infill setting.

bubnikv, 2020-12-10:
What you are proposing is similar to Cura's gradual infill.

livingkurt, 2020-12-12:
Here are some examples from inside of PrusaSlicer of how I do it manually.

How the modifiers look. It was pretty tedious to get it lined up just right so it would only effect one or 2 layers with the modifier 
[image]

How the effect the Slice.
[image]
[image]

This print turned out great with these settings. Maximized speed while maintaining a good finish!
15% Infill Everywhere and 50% Infill 1 or 2 layers below large surface areas

lukasradek, 2021-01-04:
> This print turned out great with these settings. Maximized speed while maintaining a good finish!
> 15% Infill Everywhere and 50% Infill 1 or 2 layers below large surface areas

Thanks for the example.
Does the change from 15% to 50% infill work? It feels like the 50% is bridging quite a lot above the 15%.

livingkurt, 2021-01-04:
You are welcome! It actually works great! I didn't notice any issues bridging with higher infill over lower. And now referencing back to my .3mf file I actually used 5% infill for everything and 50% for below large surface area (sorry about that). Here is how it turned out! It's a cosmetic holder for my girlfriend.
[image]
[image]

livingkurt, 2022-04-07:
Sorry this thread got lost in my inbox. Yes so far the 50 infill above 15 as worked great!. Im sure that 30 would be fine as well instead of 50.
Maybe even a gradual increase of infill density as the large flat surface gets closer, 
15%, 20%, 25% etc, up to a sufficient infill for a clean surface to take place.

stepikovo, 2022-04-11:
@livingkurt have you tried the **Support Cubic** pattern? Unfortunately, you haven't provided us with 3MF nor STL so we can't test it on your model, so please do and let us know if this is solving the issue.

And yes, when doing it manually, 30 % is fine, the same process is actually used on some of our productions parts, 10 % for speed and then the last 3 layers with 20 % to support the top layers.

livingkurt, 2022-04-24:
I have not tried the Support Cubic setting yet, but I am using the Cubic setting currently.
Here are 2 .3mf file as examples

One with the technique manually implemented and the second without it implemented.

Yes the goal of this feature would be to choose a single infill setting as you would normally and behind the scenes the slicer will automatically add in a couple layers of more dense infill right before the expanse of flat surface.
A gradient of infill densities would work the best I believe, but 1 or 2 layers is fine as well.

Thank you for your patiences with this, this email thread is getting lost in my inbox.

Jan-Soustruznik, 2025-03-19:
It should already be implemented—enhancement of the solid infill in PrusaSlicer 2.6.0-Alpha5. 

changelog: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5

I closed the issue as a duplicate.

closed 2025-03-19 by Jan-Soustruznik