prusa3d/PrusaSlicer #4305: [Feature request] Paint support

Animo78, 2020-05-26:
### Version
2.#.#

### Operating system type + version
n.v.t.

### 3D printer brand / version + firmware version (if known)
n.v.t.

_Is this a new feature request?_ Yes

I would like an option where you can do a Pre-slice to see where there is overhanging or support needed (Example turning red) and that you can paint/touch the parts where you can add support, instead of placing blocks.
Some slicers are starting with this feature as well.

pedjas, 2020-05-26:
This is actually good idea. I found it odd that althrough slicer knows what areas are overhangs I have to struggle to set block to cover the area.

It would be great to have an option to simply click on overhang I want to handle and let Slicer set blocks apropriately

Animo78, 2020-05-26:
I had several times I needed very little support.
But when I turn it on, it's flooded with support.
Then I started using those support blocks, but those aren't that accurate, as I would like

lukasmatena, 2020-05-26:
This feature is currently being worked on for 2.2.0 release.

rtyr, 2020-10-17:
Implemented in 2.3.0-alpha1. Closing.

closed 2020-10-17 by rtyr