prusa3d/PrusaSlicer #4308: [Bug] Bridging angle of 0° is 90°

qwewer0, 2020-05-26:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10, Version 1909

### Behavior
Bridging angle of 0° is in 90° to what it suppose to be.

Bridging angle of 0°
[image]

Bridging angle of 1°
[image]

Bridging angle of 15°
[image]

Bridging angle of 90°
[image]

neophyl, 2020-05-26:
You could just read the tooltip for that setting.  0 is for automatic , if you want 0 then use 180

qwewer0, 2020-05-26:
My bad, closing this.

closed 2020-05-26 by qwewer0