prusa3d/PrusaSlicer #4309: Colour change bar issue

pepperm2, 2020-05-26:
2.2.0 PrusaSlicer 
Windows 10 Operating system 
Prusa i3 MK3S 3D printer brand

With the Colour change bar You can change the colour displayed with all colour changes, on the bar, except for the base colour which always is set to Orange. It would be nice it the base or bottom colour display could be changed so that a more accurate representation or the object being printed can be seen in Prucer Slicer.

I suspect that this could be a new feature request but actually I think that it is an oversight with current functionality.

neophyl, 2020-05-27:
Isn't the base colour used whatever you have set for that filament profile ?  So you could just set your filament colour to whatever you want ?

rtyr, 2020-05-27:
It is like neophyl said, "base" color is taken from filament profile.

[image]

pepperm2, 2020-05-27:
Yes but that base colour in the filament profile gets reset to orange, it isn’t saved with the project.

rtyr, 2020-05-27:
It is saved with the project. You have to open project file by "File -> Open project".

rtyr, 2020-05-27:
Everything was answered. I don't see any issue here so I believe we can close this.

closed 2020-05-27 by rtyr