prusa3d/PrusaSlicer #4310: individual support structure

JPluess, 2020-05-26:
V. 2.2.0

How can i remove individual support structures.
Can support structure be selected with the mouse and then deleted?
Is there a checkbox print support structure only inside part or only outside part. (For some parts i cant remove it inside, like closed air ducts for a fan)

Thanks

rtyr, 2020-05-27:
You can use support enforcers and support blockers for that purpose.

More information is here https://help.prusa3d.com/en/article/modifier-meshes-custom-supports-and-other-magic_114258#custom-supports. Feel free to contact our tech-support on live-chat. Closing.

closed 2020-05-27 by rtyr

JPluess, 2020-05-28:
Well that feels not intuitive.
when stepping through the model i would like to klick on a support and remove it, or even modify it separately. here is an image. Support under the mouse should be recognized and selected more intuitively.

[image]

rtyr, 2020-05-29:
We are working on some significant improvements for next release. See #4305.