prusa3d/PrusaSlicer #4311: Possible bug in rotation or settling code?

cjshaker, 2020-05-27:
[Readme.txt](https://github.com/prusa3d/PrusaSlicer/files/4685723/Readme.txt)
[SystemInformation.txt](https://github.com/prusa3d/PrusaSlicer/files/4685725/SystemInformation.txt)

### Version
PrusaSlicer-2.2.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 (build 18363), 64-bit edition
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Stock Prusa i3 Original MK3S, firmware version 3.9.0

### Behavior
*When I rotate the part 90 degrees, it does NOT settle onto print bed correctly
* Load STL file FT3D_Stand_cjshaker_V4.stl from https://github.com/cjshaker/PrusaSlicer-2.2.0-problem. If you load this file into PrusaSlicer-2.2.0, the part will be on its side on the print bed.
If you rotate the part forward 90 degrees, so that the bottom of the part is on the print bed, it doesn't settle correctly onto the print bed. I've recreated this problem multiple times. I tried using the 'settle' command, and it does NOT fix it. You can tell the software thinks it is not settled that because the Brim goes under the middle of the part, and it prints 'support' material under most of it.
If I use the Windows 10 3D Builder to do the 90 degree rotation instead, and then import the updated STL file into PrusaSlicer, everything works.

[FT3D_Stand_cjshaker_V4.zip](https://github.com/prusa3d/PrusaSlicer/files/4685779/FT3D_Stand_cjshaker_V4.zip)

[image]
[image]

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

JohnnyDeer, 2020-05-27:
Hi @cjshaker , unfortunately i'm not able to reproduce your issue. Rotation and "place on face" works right when i'm trying to reproduce it.

JohnnyDeer, 2020-05-27:
We did find out that in your 3mf, that model was rotated by 0.27° before 90° rotation. When you switch Object manipulation tab from World coordinates to Local, you will see X rotation is 90.27°. All rotations are counted together, that mean, at the beggin you may accidently rotate object by 0.27°. Please try function Place o face, that should solve it.

[image]

cjshaker, 2021-01-08:
Running into the same problem again. Will try that advice.
I used to use the 'command line'? to rotate a part around a particular axis. I liked that better than the graphical way

cjshaker, 2021-01-08:
Yes, switching to Local Coordinates helped me to make sure that the pieces ARE actually settled (placed on face).

I contend that this behavior is surprising to the average user, and is probably NOT what most users want to happen when they click on the 'settle' button.

bubnikv, 2021-01-08:
What is the 'settle' button?

How did you create the situation where the object was rotated by 90.27, not by 90?

bubnikv, 2021-01-08:
I quite do not understand your issue, however we have fixed some numerical issue in rotation code, so you may find out that https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc3 works better.

cjshaker, 2021-01-08:
Sorry, I call your "place on face" button "settle".

It got that way because I used the graphical rotation tool in an attempt 
to turn it 90 degrees. It didn't snap to 90 degrees.

Chris Shaker

bubnikv, 2021-01-11:
> It got that way because I used the graphical rotation tool in an attempt to turn it 90 degrees. It didn't snap to 90 degrees.

But then it is user's issue.

The "Place on face" had numeric issues, but that should be fixed in PrusaSlicer 2.3.0-rc3.

cjshaker, 2021-01-12:
I use Microsoft's free "3D Builder". It's "Place on face" is called 
"settle". And it just works. If Slic3R is too much
of a pain in the ass, I can just use "3D Builder" instead.
Chris Shaker

rtyr, 2021-02-18:
Initial issue resolved. Place on face was improved in 2.3.0. Closing.

closed 2021-02-18 by rtyr