prusa3d/PrusaSlicer #4312: [Optimization] Pause Feature in PrusaSlicer

blubfix, 2020-05-27:
### Version
PrusaSlicer-2.2.0+-202003211132

### OS
Macintosh OS 10.15.4

###Model
Prusa Mk3s

###Optimization of pause feature in PrusaSlicer 

I would suggest to add a reheating to the Pause feature.
I added some commands to reheat the Nozzle and extrude a bit since the temperature resets when you drop a M601.

M601; Pause the Print
M104 S195 ; set extruder temp
M109 S195 ; set extruder temp and wait 
G1 E3; Extrude material to guarantee continuous printing 

I would love to see this feature in PrusaSlicer since the Pause feature is great for inserting things into the print.

Thx 😄 
Greetings
Blubfix

rtyr, 2020-05-27:
Your printer doesn't reheat on resume automatically? It should be done by firmware.

rtyr, 2020-05-27:
I just tested it with current FW3.9.0 and it behaves like I mentioned. Hotend temperature is set to 0C on pause (intended behavior) and automatically set to target temperature on resume. Anyway, this is completely handled by printer firmware, not by PrusaSlicer. Closing here. You can discuss this further at prusa3d/Prusa-Firmware, here is something related https://github.com/prusa3d/Prusa-Firmware/issues/2460

closed 2020-05-27 by rtyr