prusa3d/PrusaSlicer #4313: Linking errors wxwidgets static build

JoduWei, 2020-05-27:
Version 2.3.0-alpha0 

I tried it with ubuntu linux, Windows (visual studio 2019) and a fresh Ubuntu installation.

deps-build:
cmake .. -DDESTDIR=/root/PrusaSlicer-version_2.3.0-alpha0/
prusaslicer:
cmake .. -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=/root/PrusaSlicer-version_2.3.0-alpha0/usr/local

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

```[100%] Linking CXX executable prusa-slicer
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `Slic3r::GUI::GUI_App::run_wizard(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage)':
GUI_App.cpp:(.text+0x2bc3): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x2bf3): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x2bfa): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `Slic3r::GUI::GUI_App::add_config_menu(wxMenuBar*)':
GUI_App.cpp:(.text+0x7129): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x7155): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x715c): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `Slic3r::GUI::GUI_App::set_label_clr_modified(wxColour const&)':
GUI_App.cpp:(.text+0x835c): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x8388): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x838f): undefined reference to `wxTrapInAssert'
/usr/bin/ld: GUI_App.cpp:(.text+0x83cf): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x83fb): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x8402): undefined reference to `wxTrapInAssert'
/usr/bin/ld: GUI_App.cpp:(.text+0x843f): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x846b): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x8472): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `Slic3r::GUI::GUI_App::set_label_clr_sys(wxColour const&)':
GUI_App.cpp:(.text+0x887c): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x88a8): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x88af): undefined reference to `wxTrapInAssert'
/usr/bin/ld: GUI_App.cpp:(.text+0x88ef): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x891b): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x8922): undefined reference to `wxTrapInAssert'
/usr/bin/ld: GUI_App.cpp:(.text+0x895f): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x898b): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text+0x8992): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `Slic3r::GUI::GUI_App::on_init_inner()':
GUI_App.cpp:(.text+0x99da): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text+0x9afb): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, wxString const&)'
/usr/bin/ld: GUI_App.cpp:(.text+0x9b02): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `wxArgNormalizerWchar<wxString const&>::wxArgNormalizerWchar(wxString const&, wxFormatString const*, unsigned int)':
GUI_App.cpp:(.text._ZN20wxArgNormalizerWcharIRK8wxStringEC2ES2_PK14wxFormatStringj[_ZN20wxArgNormalizerWcharIRK8wxStringEC5ES2_PK14wxFormatStringj]+0x22): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text._ZN20wxArgNormalizerWcharIRK8wxStringEC2ES2_PK14wxFormatStringj[_ZN20wxArgNormalizerWcharIRK8wxStringEC5ES2_PK14wxFormatStringj]+0x4e): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text._ZN20wxArgNormalizerWcharIRK8wxStringEC2ES2_PK14wxFormatStringj[_ZN20wxArgNormalizerWcharIRK8wxStringEC5ES2_PK14wxFormatStringj]+0x55): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `Slic3r::GUI::main(int, char**)':
GUI_App.cpp:(.text.startup+0xb): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `wxArgNormalizerWchar<char const*>::wxArgNormalizerWchar(char const*, wxFormatString const*, unsigned int)':
GUI_App.cpp:(.text._ZN20wxArgNormalizerWcharIPKcEC2ES1_PK14wxFormatStringj[_ZN20wxArgNormalizerWcharIPKcEC5ES1_PK14wxFormatStringj]+0x89): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text._ZN20wxArgNormalizerWcharIPKcEC2ES1_PK14wxFormatStringj[_ZN20wxArgNormalizerWcharIPKcEC5ES1_PK14wxFormatStringj]+0xb5): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text._ZN20wxArgNormalizerWcharIPKcEC2ES1_PK14wxFormatStringj[_ZN20wxArgNormalizerWcharIPKcEC5ES1_PK14wxFormatStringj]+0xbc): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `wxString wxString::Format<char const*>(wxFormatString const&, char const*)':
GUI_App.cpp:(.text._ZN8wxString6FormatIPKcEES_RK14wxFormatStringT_[_ZN8wxString6FormatIPKcEES_RK14wxFormatStringT_]+0x7e): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text._ZN8wxString6FormatIPKcEES_RK14wxFormatStringT_[_ZN8wxString6FormatIPKcEES_RK14wxFormatStringT_]+0xaa): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text._ZN8wxString6FormatIPKcEES_RK14wxFormatStringT_[_ZN8wxString6FormatIPKcEES_RK14wxFormatStringT_]+0xb1): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_App.cpp.o): in function `void wxLogger::Log<char const*>(wxFormatString const&, char const*)':
GUI_App.cpp:(.text._ZN8wxLogger3LogIPKcEEvRK14wxFormatStringT_[_ZN8wxLogger3LogIPKcEEvRK14wxFormatStringT_]+0x7e): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: GUI_App.cpp:(.text._ZN8wxLogger3LogIPKcEEvRK14wxFormatStringT_[_ZN8wxLogger3LogIPKcEEvRK14wxFormatStringT_]+0xaa): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: GUI_App.cpp:(.text._ZN8wxLogger3LogIPKcEEvRK14wxFormatStringT_[_ZN8wxLogger3LogIPKcEEvRK14wxFormatStringT_]+0xb1): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(MainFrame.cpp.o): in function `Slic3r::GUI::MainFrame::on_presets_changed(Slic3r::GUI::SimpleEvent&)':
MainFrame.cpp:(.text+0x2593): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: MainFrame.cpp:(.text+0x25bb): undefined reference to `wxTheAssertHandler'
/usr/bin/ld: MainFrame.cpp:(.text+0x25e3): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: MainFrame.cpp:(.text+0x25ea): undefined reference to `wxTrapInAssert'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(MainFrame.cpp.o): in function `Slic3r::GUI::MainFrame::on_value_changed(wxCommandEvent&)':
[... 900 lines cut ...]
/usr/bin/ld: AboutDialog.cpp:(.text._ZN20wxEventFunctorMethodI14wxEventTypeTagI14wxCommandEventEN6Slic3r3GUI16CopyrightsDialogE7wxEventS5_EclEP12wxEvtHandlerRS6_[_ZN20wxEventFunctorMethodI14wxEventTypeTagI14wxCommandEventEN6Slic3r3GUI16CopyrightsDialogE7wxEventS5_EclEP12wxEvtHandlerRS6_]+0x6b): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
[... cut ...]
/usr/bin/ld: AboutDialog.cpp:(.text._ZN8wxString6FormatIhhhEES_RK14wxFormatStringT_T0_T1_[_ZN8wxString6FormatIhhhEES_RK14wxFormatStringT_T0_T1_]+0x15d): undefined reference to `wxOnAssert(char const*, int, char const*, char const*, char const*)'
/usr/bin/ld: AboutDialog.cpp:(.text._ZN8wxString6FormatIhhhEES_RK14wxFormatStringT_T0_T1_[_ZN8wxString6FormatIhhhEES_RK14wxFormatStringT_T0_T1_]+0x164): undefined reference to `wxTrapInAssert'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/PrusaSlicer.dir/build.make:135: src/prusa-slicer] Error 1
make[1]: *** [CMakeFiles/Makefile2:697: src/CMakeFiles/PrusaSlicer.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

lukasmatena, 2020-05-27:
There is apparently something wrong when linking PS to wxWidgets. Could you please give us exact git revision you're trying to compile and the cmake output for both builds?

JoduWei, 2020-05-27:
@lukasmatena 
I am trying to compile the zip archive in releases.
**cmake output deps:**
`- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PrusaSlicer deps DESTDIR: /root/PrusaSlicer-version_2.3.0-alpha0
-- PrusaSlicer deps debug build: ON
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /root/PrusaSlicer-version_2.3.0-alpha0/deps/build`
**cmake output prusaslicer:**
```-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: 1
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: 1
-- CMAKE_PREFIX_PATH: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/ (from cache or command line)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found DBUS: /usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include  
-- Found Boost 1.70.0 at /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS system;filesystem;thread;log;locale;regex;chrono;atomic;date_time
-- Found boost_headers 1.70.0 at /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/boost_headers-1.70.0
-- Found boost_system 1.70.0 at /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/boost_system-1.70.0
--   libboost_system.a
-- Adding boost_system dependencies: headers
-- Found boost_filesystem 1.70.0 at /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/boost_filesystem-1.70.0
[... 70 lines cut ...]
-- Found Boost: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found version "1.70.0") found components: iostreams system 
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
-- OpenVDB libraries: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libopenvdb.a
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- IGL NOT found, using bundled version...
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found GMP: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libgmp.a  
-- Found MPFR: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libmpfr.a  
-- Found Boost 1.70.0 at /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED
-- Found Boost: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.48")  
-- Boost include dirs: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/include
-- Boost libraries:    Boost::system;Boost::filesystem;Boost::thread;Boost::log;Boost::locale;Boost::regex;Boost::chrono;Boost::atomic;Boost::date_time
-- Using gcc version 4 or later. Adding -frounding-math
-- wx-config path: /root/PrusaSlicer-version_2.3.0-alpha0/usr/local/bin/wx-config
-- wx libs: -L/root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib;-pthread;/root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libwx_gtk2u_gl-3.1.a;/root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libwx_gtk2u_html-3.1.a;/root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libwx_gtk2u_core-3.1.a;/root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libwx_baseu-3.1.a;libSM.so;libICE.so;libX11.so;libXext.so;libglib-2.0.so;libgobject-2.0.so;libatk-1.0.so;libgio-2.0.so;libgthread-2.0.so;libgmodule-2.0.so;libgdk_pixbuf-2.0.so;libcairo.so;libpango-1.0.so;libpangocairo-1.0.so;libpangoft2-1.0.so;libpangoxft-1.0.so;libgdk-x11-2.0.so;libgtk-x11-2.0.so;libGLX.so;libGLU.so;libSM.so;libICE.so;libX11.so;libXext.so;libglib-2.0.so;libgobject-2.0.so;libatk-1.0.so;libgio-2.0.so;libgthread-2.0.so;libgmodule-2.0.so;libgdk_pixbuf-2.0.so;libcairo.so;libpango-1.0.so;libpangocairo-1.0.s [... cut ...] o;libgmodule-2.0.so;libgdk_pixbuf-2.0.so;libcairo.so;libpango-1.0.so;libpangocairo-1.0.so;libpangoft2-1.0.so;libpangoxft-1.0.so;libgdk-x11-2.0.so;libgtk-x11-2.0.so;libz.so;-lwxregexu-3.1;libc.so;-lpthread;dl;/usr/lib/x86_64-linux-gnu/libpng.so;/usr/lib/x86_64-linux-gnu/libz.so;/root/PrusaSlicer-version_2.3.0-alpha0/usr/local/lib/libexpat.a;/usr/lib/x86_64-linux-gnu/librt.so
-- Configuring done
-- Generating done
-- Build files have been written to: /root/PrusaSlicer-version_2.3.0-alpha0/build```

JoduWei, 2020-05-27:
```ls ../usr/local/lib/
cmake/                       libboost_locale.a            libcurl.la                   libgtest_main.a              libmpfr.a                    libqhull_p.so.7.3.2          libqhullstatic_r.a           libwx_gtk2u_adv-3.1.a        libwx_gtk2u_richtext-3.1.a
engines-1.1/                 libboost_log.a               libexpat.a                   libHalf-2_4.a                libmpfr.la                   libqhull_r.so                libssl.a                     libwx_gtk2u_aui-3.1.a        libwx_gtk2u_stc-3.1.a
libblosc.a                   libboost_log_setup.a         libglew.a                    libIex-2_4.a                 libnlopt.a                   libqhull_r.so.7              libtbbmalloc_proxy_static.a  libwx_gtk2u_core-3.1.a       libwx_gtk2u_xrc-3.1.a
libboost_atomic.a            libboost_regex.a             libgmp.a                     libIexMath-2_4.a             libopencsg.a                 libqhull_r.so.7.3.2          libtbbmalloc_static.a        libwx_gtk2u_gl-3.1.a         libwxjpeg-3.1.a
libboost_chrono.a            libboost_system.a            libgmp.la                    libIlmImf-2_4.a              libopenvdb.a                 libqhull.so                  libtbb_static.a              libwx_gtk2u_html-3.1.a       libwxregexu-3.1.a
libboost_date_time.a         libboost_thread.a            libgmpxx.a                   libIlmImfUtil-2_4.a          libqhullcpp.a                libqhull.so.7                libwx_baseu-3.1.a            libwx_gtk2u_propgrid-3.1.a   libwxscintilla-3.1.a
libboost_filesystem.a        libcrypto.a                  libgmpxx.la                  libIlmThread-2_4.a           libqhull_p.so                libqhull.so.7.3.2            libwx_baseu_net-3.1.a        libwx_gtk2u_qa-3.1.a         pkgconfig/
libboost_iostreams.a         libcurl.a                    libgtest.a                   libImath-2_4.a               libqhull_p.so.7              libqhullstatic.a             libwx_baseu_xml-3.1.a        libwx_gtk2u_ribbon-3.1.a     wx/```

HaythamB, 2020-06-18:
Getting a similar error on a clean VS2019 installation on Windows 10, latest git for Windows installed. Any news on how to fix this?

HaythamB, 2020-06-18:
Also just ran another clean install of the files. 2.2.0 release works fine and runs. 2.3 alpha from today's date won't run, and I suspect it might be the following line that has they key to the fix:

`27>   Creating library C:/PrusaSlicer/psa/build/src/Debug/PrusaSlicer.lib and object C:/PrusaSlicer/psa/build/src/Debug/PrusaSlicer.exp
27>LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library
27>libslic3r_gui.lib(2DBed.obj) : error LNK2001: unresolved external symbol "void __cdecl wxOnAssert(char const *,int,char const *,char const *,char const *)" (?wxOnAssert@@YAXPEBDH000@Z)`

timemaster5, 2020-06-21:
the same on Catalina, do we have any workaround?

soswow, 2020-07-23:
I am getting 
```
[ 98%] Linking CXX executable prusa-slicer
Undefined symbols for architecture x86_64:
  "wxOnAssert(char const*, int, char const*, char const*, wchar_t const*)", referenced from:
      wxBoxSizer::wxBoxSizer(int) in liblibslic3r_gui.a(Plater.cpp.o)
      wxNavigationEnabled<wxWindow>::wxNavigationEnabled() in liblibslic3r_gui.a(Plater.cpp.o)
      wxBoxSizer::wxBoxSizer(int) in liblibslic3r_gui.a(MainFrame.cpp.o)
      Slic3r::GUI::MainFrame::add_to_recent_projects(wxString const&) in liblibslic3r_gui.a(MainFrame.cpp.o)
      wxNavigationEnabled<wxWindow>::wxNavigationEnabled() in liblibslic3r_gui.a(MainFrame.cpp.o)
      wxEventFunctorFunctor<wxEventTypeTag<wxCloseEvent>, Slic3r::GUI::MainFrame::MainFrame()::$_1>::operator()(wxEvtHandler*, wxEvent&) in liblibslic3r_gui.a(MainFrame.cpp.o)
      wxEventFunctorFunctor<wxEventTypeTag<wxCommandEvent>, Slic3r::GUI::MainFrame::init_menubar()::$_9>::operator()(wxEvtHandler*, wxEvent&) in liblibslic3r_gui.a(MainFrame.cpp.o)
      ...
  "wxOnAssert(char const*, int, char const*, char const*, wxString const&)", referenced from:
      Slic3r::GUI::GUI_App::on_init_inner() in liblibslic3r_gui.a(GUI_App.cpp.o)
  "wxOnAssert(char const*, int, char const*, char const*, char const*)", referenced from:
      Slic3r::GUI::OtherInstanceMessageHandler::handle_message(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in liblibslic3r_gui.a(InstanceCheck.cpp.o)
      Slic3r::GUI::OptionsGroup::hide_labels() in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::Sidebar::show_info_sizer() in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<double, double, double>(wxFormatString const&, double, double, double) in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<int, int>(wxFormatString const&, int, int) in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<int, int, int, int, int, int>(wxFormatString const&, int, int, int, int, int, int) in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::Sidebar::update_sliced_info_sizer() in liblibslic3r_gui.a(Plater.cpp.o)
      ...
  "_wxTheAssertHandler", referenced from:
      Slic3r::GUI::OtherInstanceMessageHandler::handle_message(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in liblibslic3r_gui.a(InstanceCheck.cpp.o)
      wxBoxSizer::wxBoxSizer(int) in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::OptionsGroup::hide_labels() in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::Sidebar::show_info_sizer() in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<double, double, double>(wxFormatString const&, double, double, double) in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<int, int>(wxFormatString const&, int, int) in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<int, int, int, int, int, int>(wxFormatString const&, int, int, int, int, int, int) in liblibslic3r_gui.a(Plater.cpp.o)
      ...
  "_wxTrapInAssert", referenced from:
      Slic3r::GUI::OtherInstanceMessageHandler::handle_message(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in liblibslic3r_gui.a(InstanceCheck.cpp.o)
      wxBoxSizer::wxBoxSizer(int) in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::OptionsGroup::hide_labels() in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::Sidebar::show_info_sizer() in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<double, double, double>(wxFormatString const&, double, double, double) in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<int, int>(wxFormatString const&, int, int) in liblibslic3r_gui.a(Plater.cpp.o)
      wxString wxString::Format<int, int, int, int, int, int>(wxFormatString const&, int, int, int, int, int, int) in liblibslic3r_gui.a(Plater.cpp.o)
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/prusa-slicer] Error 1
make[1]: *** [src/CMakeFiles/PrusaSlicer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
Not sure if the same. 
I am at `5980d46e4`

bubnikv, 2021-09-25:
We don't have resources to support self-compilation issues.

We have updated our compilation guides. You may find some advice there.
https://github.com/prusa3d/PrusaSlicer/wiki/Building-PrusaSlicer

closed 2021-09-25 by bubnikv