prusa3d/PrusaSlicer #4315: Home axes before wait for temperatures

espr14, 2020-05-27:
### Version
PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
MINI 4.0.5
i3 MK3S 3.9.0

### Behavior
Default profile:
1. set temperatures
2. wait for temperatures
3. home
4. MBL

Proposed profile:
1. set temperatures
2. home
3. wait for temperatures
4. MBL

_Is this a new feature request?_ **YES**

#### Project File (.3MF) where problem occurs
N/A

lukasmatena, 2020-05-27:
What problem does it solve?

espr14, 2020-05-27:
Saves time to get nozzle to [0,0,0]. Move and heat in parallel. If the nozzle is far away after previous print, it takes few seconds to get home.

foreachthing, 2020-05-27:
You can do exactly that with custom (printer - start-code) gcode.

espr14, 2020-05-27:
It's not about me. It's about all i3 MK2, 2.5, 3, S, MMU1, MMU2 and MINI and future printers and MMUs, for all nozzles for all users using default or deviating from default.