prusa3d/PrusaSlicer #4316: Slicer Crashes all the time

philipphagen87, 2020-05-27:
### Version: 2.2.0+win64
also testet the inofficial version 2.3.0-alpha1

### Operating system type + version: 10.0.18362 Build 18362

### 3D printer brand / version + firmware version: Prusa i3 MK3S/Firmware 3.9.0

### Behavior
After starting the application it takes same second about 10-20sec and then the application shut down - error message: (sorry for the german language) ;-)

First error:
[image]

second error:
[image]

is there any logfile implemented to find some issue?

lukasmatena, 2020-05-27:
You can (and we would appreciate it) run PrusaSlicer from the command line as
`prusa-slicer-console.exe --loglevel=9` and see if there's something interesting before it crashes.

You can also run it using software renderer to rule out issues with hardware acceleration:
`prusa-slicer-console.exe --sw-renderer --loglevel=9`

There are more issues like this, they may not have all the same cause. See for example #3177 or #3374 (btw. that guy is also running in Deutch, and although a connection is a bit far fetched, you can try switching to English and see).

Please, report whatever you find out. It would be nice to have these spurious Windows crashes fixed.

philipphagen87, 2020-05-27:
FYI yesterday everything worked well, it was running for about 3-4 hours...

Here is the Log...

`C:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer-console.exe --sw-renderer --loglevel=9
Loading MESA OpenGL library: C:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
[2020-05-27 18:25:43.532924] [0x000033c0] [info]    Checking if indices need to be installed from resources...
[2020-05-27 18:25:43.537906] [0x000033c0] [trace]   System language detected (user locales and such): de_AT
[2020-05-27 18:25:43.538867] [0x000033c0] [trace]   Best translation language detected (may be different from user locales): de_DE
[2020-05-27 18:25:43.538867] [0x000033c0] [trace]   Switching wxLocales to de_DE
[2020-05-27 18:25:44.673334] [0x000033c0] [info]    Undo / Redo snapshot taken: Neues Projekt, Undo / Redo stack memory: 0MB WorkingSet: 39MB; PrivateBytes: 18MB; Pagefile(peak): 18MB(19MB)
[2020-05-27 18:25:44.745080] [0x00004c2c] [trace]   Mouse3DController::connect_device() - enumerating HID devices.
[2020-05-27 18:25:51.399911] [0x000033c0] [debug]   TriangleMesh::repair() started
[2020-05-27 18:25:51.407036] [0x000033c0] [debug]   TriangleMesh::repair() finished
[2020-05-27 18:25:51.409886] [0x000033c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-05-27 18:25:51.415889] [0x000033c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-05-27 18:25:51.422848] [0x000033c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-05-27 18:25:51.428882] [0x000033c0] [debug]   TriangleMesh::repair() started
[2020-05-27 18:25:51.434902] [0x000033c0] [debug]   TriangleMesh::repair() finished
[2020-05-27 18:25:51.440803] [0x000033c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-05-27 18:25:51.443792] [0x000033c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-05-27 18:25:51.449777] [0x000033c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-05-27 18:25:52.514928] [0x000033c0] [info]    Checking for cached configuration updates...
[2020-05-27 18:25:52.524901] [0x000033c0] [info]    Confing bundle not installed for vendor BIBO, skipping:
[2020-05-27 18:25:52.526895] [0x000033c0] [info]    Confing bundle not installed for vendor Creality, skipping:
[2020-05-27 18:25:52.531882] [0x000033c0] [info]    Confing bundle not installed for vendor LulzBot, skipping:
[2020-05-27 18:25:52.554855] [0x000033c0] [debug]   Vendor: Prusa Research, version installed: 1.1.6, version cached: 1.1.6
[2020-05-27 18:25:52.560805] [0x000033c0] [info]    No configuration updates available.
[2020-05-27 18:25:52.606717] [0x00004d68] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
*   Trying 185.115.1.124...
* TCP_NODELAY set
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* schannel: SSL/TLS connection with files.prusa3d.com port 443 (step 1/3)
* schannel: checking server certificate revocation
* schannel: ALPN, offering http/1.1
* schannel: sending initial handshake data: sending 197 bytes...
* schannel: sent initial handshake data: sent 197 bytes
* schannel: SSL/TLS connection with files.prusa3d.com port 443 (step 2/3)
* schannel: failed to receive handshake, need more data
* schannel: SSL/TLS connection with files.prusa3d.com port 443 (step 2/3)
* schannel: encrypted data got 3062
* schannel: encrypted data buffer: offset 3062 length 4096
* schannel: sending next handshake data: sending 158 bytes...
* schannel: SSL/TLS connection with files.prusa3d.com port 443 (step 2/3)
* schannel: encrypted data got 51
* schannel: encrypted data buffer: offset 51 length 4096
* schannel: SSL/TLS handshake complete
* schannel: SSL/TLS connection with files.prusa3d.com port 443 (step 3/3)
* schannel: ALPN, server accepted to use http/1.1
* schannel: stored credential handle in session cache
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.2.0+win64
Accept: */*

* schannel: client wants to read 16384 bytes
* schannel: encdata_buffer resized 17408
* schannel: encrypted data buffer: offset 0 length 17408
* schannel: encrypted data got 242
* schannel: encrypted data buffer: offset 242 length 17408
* schannel: decrypted data length: 213
* schannel: decrypted data added: 213
* schannel: decrypted data cached: offset 213 length 16384
* schannel: encrypted data buffer: offset 0 length 17408
* schannel: decrypted data buffer: offset 213 length 16384
* schannel: schannel_recv cleanup
* schannel: decrypted data returned 213
* schannel: decrypted data buffer: offset 0 length 16384
< HTTP/1.1 200 OK
[... 64 lines cut ...]
* schannel: encdata_buffer resized 17408
* schannel: encrypted data buffer: offset 0 length 17408
* schannel: encrypted data got 7551
* schannel: encrypted data buffer: offset 7551 length 17408
* schannel: decrypted data length: 7522
* schannel: decrypted data added: 7522
* schannel: decrypted data cached: offset 7522 length 16384
* schannel: encrypted data buffer: offset 0 length 17408
* schannel: decrypted data buffer: offset 7522 length 16384
* schannel: schannel_recv cleanup
* schannel: decrypted data returned 7522
* schannel: decrypted data buffer: offset 0 length 16384
< HTTP/1.1 200 OK
< Server: nginx
< Date: Wed, 27 May 2020 16:25:53 GMT
< Content-Length: 7309
< Connection: keep-alive
< Last-Modified: Wed, 27 May 2020 16:00:17 GMT
< ETag: "1c8d-5a6a34eed5853"
< Accept-Ranges: bytes
<
* Connection #1 to host files.prusa3d.com left intact
[2020-05-27 18:25:53.545549] [0x00004d68] [debug]   Got index for vendor: Prusa Research: current version: 1.1.6, recommended version: 1.1.6

C:\Program Files\Prusa3D\PrusaSlicer>`

philipphagen87, 2020-05-29:
Eventlog of Windows:

German (original)
Name der fehlerhaften Anwendung: prusa-slicer.exe, Version: 2.2.0.0, Zeitstempel: 0x5e75ff9e
Name des fehlerhaften Moduls: PrusaSlicer.dll, Version: 0.0.0.0, Zeitstempel: 0x5e75ff84
Ausnahmecode: 0xc000041d
Fehleroffset: 0x000000000180d303
ID des fehlerhaften Prozesses: 0x2928
Startzeit der fehlerhaften Anwendung: 0x01d635af5cb2a7ab
Pfad der fehlerhaften Anwendung: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
Pfad des fehlerhaften Moduls: C:\Program Files\Prusa3D\PrusaSlicer\PrusaSlicer.dll
Berichtskennung: c4c7aac4-2012-4320-b8d4-7559a3ad2232
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist: 

English (google translater)
Faulting application name: prusa-slicer.exe, version: 2.2.0.0, timestamp: 0x5e75ff9e
Faulting module name: PrusaSlicer.dll, version: 0.0.0.0, timestamp: 0x5e75ff84
Exception code: 0xc000041d
Error offset: 0x000000000180d303
Faulting process id: 0x2928
Faulting application start time: 0x01d635af5cb2a7ab
Faulting application path: C: \ Program Files \ Prusa3D \ PrusaSlicer \ prusa-slicer.exe
Path of the faulty module: C: \ Program Files \ Prusa3D \ PrusaSlicer \ PrusaSlicer.dll
Report ID: c4c7aac4-2012-4320-b8d4-7559a3ad2232
Full name of the faulty package:
Application ID that is relative to the bad package:

lukasmatena, 2020-06-04:
Unfortunately, I can't tell what's wrong from the log.
There is a suggestion recently posted in https://github.com/prusa3d/PrusaSlicer/issues/2939#issuecomment-638524121 (which is similar) that the problem might be related to print bed model/texture. Could you try turning that off and see if there is some difference?

If not, you can also rename your Appdata/Roaming/PrusaSlicer folder and let PS recreate it from scratch. In case that helped, we would like to see the faulty one (i.e., do not just delete it).

philipphagen87, 2020-06-09:
I tried to delete the *.ini like in post #2939 and made new standard settings, but it crashes also after seconds or sometimes minutes. Sorry for that bad news :-(

Buddinski88, 2020-07-04:
I have the same issue since a few days. Prusa slicer 2.2+ won't start anymore:

```
C:\Users\Buddy>cd C:\Program Files\Prusa3D\PrusaSlicer

C:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer-console.exe --sw-renderer --loglevel=9
Loading MESA OpenGL library: C:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
[2020-07-04 15:45:24.059610] [0x00003330] [info]    Checking if indices need to be installed from resources...
[2020-07-04 15:45:24.063615] [0x00003330] [trace]   translation_language provided by PrusaSlicer.ini: en_US
[2020-07-04 15:45:24.063615] [0x00003330] [trace]   System language detected (user locales and such): de_DE
[2020-07-04 15:45:24.065616] [0x00003330] [trace]   Best translation language detected (may be different from user locales): de_DE
[2020-07-04 15:45:24.065616] [0x00003330] [trace]   Switching wxLocales to en_US
[2020-07-04 15:45:25.113569] [0x00003330] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB WorkingSet: 39MB; PrivateBytes: 18MB; Pagefile(peak): 18MB(19MB)
[2020-07-04 15:45:25.247691] [0x00003200] [trace]   Mouse3DController::connect_device() - enumerating HID devices.
[2020-07-04 15:45:30.186188] [0x00003330] [debug]   TriangleMesh::repair() started
[2020-07-04 15:45:30.189191] [0x00003330] [debug]   TriangleMesh::repair() finished
[2020-07-04 15:45:30.190192] [0x00003330] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-07-04 15:45:30.192194] [0x00003330] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-07-04 15:45:30.194196] [0x00003330] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-07-04 15:45:30.197199] [0x00003330] [debug]   TriangleMesh::repair() started
[2020-07-04 15:45:30.199200] [0x00003330] [debug]   TriangleMesh::repair() finished
[2020-07-04 15:45:30.200201] [0x00003330] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-07-04 15:45:30.202204] [0x00003330] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-07-04 15:45:30.205206] [0x00003330] [trace]   TriangleMeshSlicer::require_shared_vertices - end
```

steckdev, 2020-07-20:
I'm having the same issue now... I was using it just fine. Then all of a sudden! The app won't launch. I have tried the install version and the stand-alone version and neither will work for me now.

KamdenB, 2020-09-19:
Im starting to have this issue too. Has anyone figured out how to fix it?

philipphagen87, 2020-11-08:
Sorry for that long time of no comments, i had no time to use Prusa... :-(

But today i worked the hole day without crashing.
[image]

I did not change anything, using on the same external montitor, same laptop (no hardware changes no os reinstall).
[image]

I hope i can help anyone :-P

philstenning, 2021-11-07:
Just happened for the second time, I'm now running windows 11 but happened on the previous windows 10 as well. I found that if I delete the C:\Users\_youUsername_\AppData\Roaming\PrusaSlicer folder it starts as if run for the first time, this is not a fix but if you just need to get something printed and use  the default settings and don't mind blowing things away it work for me.

lukasmatena, 2023-02-08:
Do you still have this issue with [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)?

lukasmatena, 2024-02-18:
I guess not. Closing.

closed 2024-02-18 by lukasmatena