prusa3d/PrusaSlicer #4318: Support Z-Distance is set to 0.2mm but is 0.8mm

BrenMcGuire-GER, 2020-05-27:
### Version
2.2.0+ Official Released 
### Operating system type + version
_MacOS 10.14_
### 3D printer brand / version + firmware version (if known)
Geeetech A20m with Marlin 2.0.5.3

### Behavior
* _Describe the problem_
The problem occurs while slicing this Project with activated support.
If the Z distance of support is set to 0mm (zero) the distance is as it should be: 0mm
If the Z distance of support is set to 0.2mm the distance isn't 0.2mm as expected but 0.6mm.
(measured by Support Interface is printed at 2.20mm, the next layer of the object is printed at 2.80mm.
Besides this, the support interface is diffent if the distance is set to 0mm (zero) as by 0.2mm. But I haven't changed anything of the support-interface settings.
[pc.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/4691226/pc.amf.zip)
[image]
[image]
[image]
[image]
[image]
[image]

* _Steps needed to reproduce the problem_
Load the Project, slice it with my or standard setting and watch the support layers at 2.20mm
* _Expected Results_
I expected that the distance between support and object is 0.2mm.
* _Actual Results_
The distance isn't 0.2mm as in the settings, its 0.6mm

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-05-27:
The bottom most surface of the overhang is printed with a bridge flow which means it is 0.4mm thick (for a nozzle with 0.4mm diameter) so it is expected behavior. You can find more discussion how PrusaSlicer (and upstream Slic3r) works with supports here #102, there are also some duplicates #2766, #926, #198, #3966, #885, #3887,... Closing.

[image]

closed 2020-05-27 by rtyr

kb-elmo, 2021-02-01:
> The bottom most surface of the overhang is printed with a bridge flow which means it is 0.4mm thick (for a nozzle with 0.4mm diameter) so it is expected behavior. You can find more discussion how [...]

This has literally nothing to do with this issue.

Even with the support material width and flow set to the exact same value as the rest of the print the contact-z distance is still wrong.

You are just distracting from the real problem here that the OP described pretty well.

And "it's expected behaviour in upstream" (which hasn't seen an update since 2018 and is just as broken as this here) is not an excuse to not fixing a very real and very annoying issue that makes this slicer pretty much unusable for prints with supports.

rtyr, 2021-02-01:
> This has literally nothing to do with this issue.

I believe you are wrong. It explains the behavior described in this issue, specifically how contact Z distance works (what may seem to be 0.6mm distance is actually 0.2mm distance, since bridge flow is considered to be a round extrusion of the same diameter as the nozzle diameter).  That being said, I can understand that it may be confusing. It is how PrusaSlicer works at the moment, so yes, it is expected behavior.

> You are just distracting from the real problem here that the OP described pretty well.
> And "it's expected behaviour in upstream" (which hasn't seen a stable update since 2018 and is just as broken as this here) is not an excuse for not fixing a very real and very annoying issue that makes this slicer pretty much unusable for prints with supports.

I don't understand, why are you so offensive. This issue was closed mainly because it is just a duplicate of other issues. The main issues on this topic are still OPENED.
We are planning to look into the supports issues globally, we are also doing some testing, comparisons, etc. Using the bridge flow on the bottom side of the supported objects is one of the topics.