prusa3d/PrusaSlicer #4319: Models taller than 2000 mm won't slice or show any preview

atmaker, 2020-05-28:
### Version
2.2.0

### Operating system type + version
Windows 10 64 bit

PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition
Total RAM size [MB]: 34,249MB
OpenGL installation
GL version:   4.6.0 NVIDIA 442.92
Vendor:       NVIDIA Corporation
Renderer:     Quadro M4000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
[... 325 lines cut ...]
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Custom / Duet 3 

### Behavior
PrusaSlicer (and Slic3r) won't slice or show any preview if the printed model is taller than 2000 mm, so there has to be a hard coded max height limit somewhere. Could you fix this or point out where to look for the issue? We would like to use PrusaSlicer during the dev of our largescale printer (current max. height around 3500 mm)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-05-28:
This is limited by the very slicing core design. The coordinates in mm are scaled by a factor of 6 and converted to integers so the internals don't have to deal with floating point pitfalls. So it internally uses nanometers, stored in 32-bit signed integer. And that overflows at 2,147,483,647, which is little over 2 meters.

The only solution is to switch to 64-bit integers, which is possible but detrimental as for memory usage. There is somewhere the same issue where someone hit this limit in x/y, so... I don't know.

atmaker, 2020-05-28:
Thanks for your quick response.Ok, that makes sense. So, this is then more of an issue with the Slic3r core?

lukasmatena, 2020-05-28:
> Slic3r core? I'll ping them

That's probably not necessary, PrusaSlicer is now detached from Slic3r, the codebases diverged significantly and the decisions are now independent.

lukasmatena, 2020-05-28:
Btw I edited my first post and added references to the other issues, if you can have a look. I see you work over email so maybe you missed the edit.

bubnikv, 2020-05-28:
@atmaker

> We would like to use PrusaSlicer during the dev of our largescale printer (current max. height around 3500 mm)

What is the build volume of your printer? What is your nozzle diameter?

I would like to point out, that PrusaSlicer (and the upstream slicer and likely any slicer) has hardcoded values, which are tuned to the common 0.4mm nozzle and similar. Therefore you will be likely better off to scale everything down before slicing and then to scale up inside the printer firmware.

lukasmatena, 2020-05-28:
I have just limited the `max_print_z` config option so PrusaSlicer at least complains loudly (2c24473). X and Y were limited already.

atmaker, 2020-05-28:
The build volume will be 1000 x 3500 mm cylindrical volume. We are going to use a pellet extruder (which will cause also other headache), but anyway the nozzle size will be 4 mm at minimum. The esteps are going to be interesting to solve.

bubnikv, 2021-09-25:
We have limited the build volume to 1.2 meters cubic. The limit is given by the 32bit resolution of our internal coordinates scaled to 1 nanometer resolution.

For your pellet machine, we recommend to just work with centimeters, centimeters per second etc... in PrusaSlicer and adjust your firmware accordingly.

closed 2021-09-25 by bubnikv

iridium13, 2023-02-10:
> @atmaker
>
>
> What is the build volume of your printer? What is your nozzle diameter?
>
> I would like to point out, that PrusaSlicer (and the upstream slicer and likely any slicer) has hardcoded values, [...]

@bubnikv I'm not following what you mean by scaling down and then scaling up in printer firmware.  Is this still applicable?  

Looking to use prusaslicer for a large printer.  The build dimensions fit within the limits, but nozzle sizes are larger than normal (0.6mm up to 10mm)

bubnikv, 2023-02-10:
Printing with a 0.6mm nozzle on such a huge build volume is not practical. More likely you will be printing with 2mm to 4mm diameter nozzles I suppose. Then it makes sense to just define your machine as a cylinder of 100cm diameter and 500cm height, nozzle diameter 0.2cm to 0.4cm and set up your machine in PrusaSlicer with those dimensions. Just substitute cm everywhere you see mm in PrusaSlicer and in your firmware.

We may change the fixed point resolution from 1 nanometer to 10 nanometers or rougher, however such change will require a huge amount of testing and fixing.

iridium13, 2023-02-10:
> Printing with a 0.6mm nozzle on such a huge build volume is not practical. More likely you will be printing with 2mm to 4mm diameter nozzles I suppose. Then it makes sense to just define your [...]

Ok I think I am following?  I don't have control over my printer firmware.  So if I work in cm in prusa slicer, then I would need to scale xyzfe commands by 10 in the output to convert them to mm.  Which is simple enough to do with quick python code.  

However... when you do this you lose some precision because prusa is exporting 3 decimals for xyz and 5 for E.  Is there a way to export more?  Based on scaling by a factor of 6 mentioned above, prusa slicer calculates everything with 6 decimals.

I guess it probably doesn't matter much though?  In terms of rounding it's +/- .005mm which is .0002"....