prusa3d/PrusaSlicer #4322: Feature Request: Add transparency settings on 3D editor view when placing support blockers

fearedspark, 2020-05-28:
### Version
Any

### Operating system type + version
Any

### Feature Requested
When adding support blockers on a complex model, adjusting the transparency of the model can help to position it properly.
I didn't see any setting for this.

neophyl, 2020-05-28:
Already requested this one some time back 🙂
https://github.com/prusa3d/PrusaSlicer/issues/3678

fearedspark, 2020-05-28:
My bad, searched tickets with the "transparent" and "transparency" keyworks, didn't think about X-Ray mode. Closing ticket.

neophyl, 2020-05-28:
Hey if you want to pile on to the one I opened and bump it up , all support welcome lol.

fearedspark, 2020-05-28:
Will see if this ticket gets some attention :)

rtyr, 2020-05-31:
Yes, but it is still a duplicate of #3678, so I will close this :-).

closed 2020-05-31 by rtyr

fearedspark, 2020-05-31:
That's fair 👍