prusa3d/PrusaSlicer #4323: Freezing and corrupting the screen when slicing the file.

bkzspam, 2020-05-28:
Hello, 
PrusaSlicer freezes trying to slice the thing. It may be an issue of graphic drivers. 
For producing it: 1. Slice 2. Wait for just 5-10 secs
It should be working good but it corrupts the screen with weird lines and then completely freezes.
Facts: It doesn't happen when using 0.6mm layer height but happens with 0.05mm layer height. 

OS: Windows 10 (build 18362) 64 Bit
Slicer version: 2.2.0 Win64
Photo of one of the behaviours, taken with a phone:
[image]
Screenshot from the computer:
[image]
The project file is here:
[project.zip](https://github.com/prusa3d/PrusaSlicer/files/4697710/project.zip)

rtyr, 2021-02-18:
This is almost certainly hardware related issue. Slicing is cpu intensive process, which likely triggers some HW issue on your computer (overheating, PSU, etc.). Slicing with 0.6mm layer height is way less cpu intensive.
Closing.

bkzspam, 2021-02-18:
This issue was triggered by only with Prusa Slicer. 
It is impossible to be PSU since it's Corsair's new 1000watt and my computer's power consumption is only about 150 watt at max.
Also it is the most obviously it's not the heat since even with overclock 400Mhz over the base frequency

bubnikv, 2021-02-19:
> Slicer version: 2.2.0 Win64

Does it happen with PrusaSlicer 2.3.0 as well?

bkzspam, 2022-02-25:
> Slicer version: 2.2.0 Win64
> […](#)

Since than lots of time passed, but it's working perfectly now. Maybe it was the Prusa slicer or the driver, don't know but this is my current favourite slicer.

Issue could be closed.

lukasmatena, 2022-02-25:
@bkzspam Thanks for the heads up.

closed 2022-02-25 by lukasmatena