prusa3d/PrusaSlicer #4327: Extruder inexplicably emits huge blob

systronix, 2020-05-30:
### Version
_Version of PrusaSlicer used goes here_
2.2.0+lunx-x64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Linux ASUS-UN62 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 18.04.4 LTS
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
i3MK2 with FW 3.2.3. Which is the most recent for MK2. It's been working great printing parts for headbands, mostly sent to Native American medical centers in the Four Corners area.

### Behavior
* _Describe the problem_
First three layers of object print OK. At a corner of third layer, a possible retraction, a huge blob is emitted when the extruder suddenly pulls a huge amount of filament through
* _Steps needed to reproduce the problem_
I have the gcode file that does this. It has now done it on three files all trying to print the same STl object, a Bear upgrade Y-axis rod holder.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
I have video of this defect happening.
I can export plate as .amf but not .3mf as you request above. Is that a typo?
Also I attach two gcode files which have this defect
[y_rod_holder.zip](https://github.com/prusa3d/PrusaSlicer/files/4706312/y_rod_holder.zip)
[Extruder_error_gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/4706313/Extruder_error_gcode.zip)

rtyr, 2020-05-30:
The issue is caused by your custom printer profile. Either re-enable "Use relative E distances" or remove G92 E0 command from "Before layer change g-code".

closed 2020-05-31 by rtyr

systronix, 2020-06-03:
Wow, thanks for a quick answer. If  "Use relative E distances" should always be checked, why is the default false/unchecked? 

I had recently gone through slicer settings and tried to make everything possible the "default" but that seems now not a good idea at all. Perhaps the "deafults" in slicer could use a review and change?

rtyr, 2020-06-03:
"Use relative E distances" can be disabled, there is no problem with that. The problem was combination of this parameter with your "Before layer change g-code" (custom gcode inserted to each layer), which expects relative E distances. This custom gcode is used in printer profiles for our printers, which are always set to relative E distances.

systronix, 2020-06-03:
OK now I understand, thanks. So when used for Prusa printers with your profiles, the tool tip for Printer Settings "Use relative E distances" doesn't warn about that and states the opposite: "Most firmwares use absolute values". That wording might want to be improved.
[image]

rtyr, 2020-06-03:
Tooltips are meant for PrusaSlicer as an aplication, it is independent from the profiles (at least at the moment). Our printers can handle both relative and absolute E positioning, we are just using relative by default.