prusa3d/PrusaSlicer #4329: Arrange only uses half of the bed when you have a large number of objects

jab416171, 2020-05-31:
### Version
2.2.0

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
* When using "Arrange" with a lot of objects, only the left side of the bed seems to be used.

#### Project File (.3MF) where problem occurs

[supercube.zip](https://github.com/prusa3d/PrusaSlicer/files/4707205/supercube.zip)

For the project file, once you delete one object (I deleted 67.stl) and then click arrange, it will use the whole bed.

As you can see in the screenshots, after removing one object, "Arrange" starts to use the whole bed. There a a few pieces that are still outside of the printing surface, but this is much better than the 25 or so that were outside before.

[image]
[image]

closed 2020-06-01 by tamasmeszaros

jab416171, 2020-06-01:
Thank you @tamasmeszaros!

tamasmeszaros, 2020-06-02:
Thanks for the report :+1:

commit 4be0e379 (2020-06-01): Workaround for items out of bed after arrange.

Fixes #4329