prusa3d/PrusaSlicer #4331: View resets to "Color Print" every time after slicing

tg73, 2020-05-31:
### Version
2.2.0+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S, 3.9.0 + 1.0.6

### Behavior
Every time I hit "Slice now", the view resets to "Color Print". I have to chose "Feature Type" from the drop down *every time*!

tg73, 2020-05-31:
It would appear that I had somehow accidentally initiated Color Print mode. I have no idea how that happened, I've never used it on purpose. It was not clear to me that I had done that. Pretty confusing.

rtyr, 2020-05-31:
It happens only when you add some color changes (at least one) in the print preview. I believe it is like that by design and it kind of makes sense, don't you think?

rtyr, 2020-05-31:
Oh I somehow missed your second comment. So we can close this.

closed 2020-05-31 by rtyr

sg000, 2020-06-11:
Actually I'd like to suggest keeping this open. Its quite annoying when you're evaluating the changes to feature types and it keeps reverting to "color print" mode. Frustrating to be exact. It should keep it to the setting I had it. Perhaps I'm evaluating speed, or flow and it keeps changing. Annoying as all get-out to have to keep changing it.