prusa3d/PrusaSlicer #4332: PrusaSlicer fails to check for updates on macOS Catalina 10.15.4

corbinschwartz, 2020-05-31:
### Version
v2.2.0+

### Operating system type + version
macOS Catalina 10.15.4

### 3D printer brand / version + firmware version (if known)
N/A (but it's an Original Prusa i3 MK3)

### Behavior
PrusaSlicer fails to check for updates on macOS Catalina 10.15.4
* _Steps needed to reproduce the problem_
Install PrusaSlicer from DMG
Open PrusaSlicer
Attempt to check for update
* _Expected Results_
Updates are checked
* _Actual Results_
Error screen with:
boost::filesystem::copy_file: No such file or directory: "/Users/corbin/Library/Application Support/PrusaSlicer/cache/PrusaResearch.idx-update", "/Users/corbin/Library/Application Support/PrusaSlicer/vendor/PrusaResearch.idx-update"

[image]

_Is this a new feature request?_
No

corbinschwartz, 2020-05-31:
I had a hunch that turned out to be correct. I gave PrusaSlicer the Full Disk Access permission and the problem went away. PrusaSlicer should request the permissions before trying to access files outside of the sandbox, or it will just error all the time.

bubnikv, 2020-06-15:
This is weird. If it was a permission issue, then you would not be able to save any of your profiles. Try making a copy of a Print profile and save it. Does it work? I suppose it will.

I believe the PrusaSlicer OSX manifests disables sand boxing, the application should have access to full disk space.

kocikdav, 2020-06-23:
Hello, thank you for reporting. Fixed at af6b24f2ecfd14c0a9e58a905d990b466e5183ad . It was causes by saving wrong paths to idx file. Im closing this issue.

closed 2020-06-23 by kocikdav

bgiot, 2020-09-21:
Ok, but what to do for fixing this right now? delete something?