prusa3d/PrusaSlicer #4333: Filement Override Retraction Length doesn't re-enable rest of options

madebyjeffrey, 2020-05-31:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Release 2004

### 3D printer brand / version + firmware version (if known)
N/A - UI Issue

### Behavior
* Setting filament overrides retraction length to 0, and undoing it doesn't re-enable the other options

* _Steps needed to reproduce the problem_
1. Select a filament
2. Go to Filament overrides
3. Enable length and set to a value that is positive and not 0
4. At this point the other options should be enabled
5. Change the length to 0
6. The other options are disabled.
7. Change the length option to a positive integer greater than 0

* _Expected Results_

The other options should be enabled.

* _Actual Results_

The other options are not enabled, they are still disabled.

Mitigation: go to another filament option and go back.

_Is this a new feature request?_

NO

rtyr, 2020-05-31:
Are you sure, that you confirmed the value in step 7 (using ENTER key)? Only way I can reproduce something similar is when I use TAB instead of ENTER. I am not sure if it is a problem or not.

rtyr, 2020-10-06:
No response. I will close this, because it works fine, as long as the retraction value is confirmed.

closed 2020-10-06 by rtyr