prusa3d/PrusaSlicer #4337: Some settings are not loaded/displayed after updating the system presets

ikarisan, 2020-06-02:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10, 1909

### 3D printer brand / version + firmware version (if known)
Hypercube Evolution + Duet3D Wifi, FW3.1

### Behavior
I have many (>15 each) individual configurations for "Print-", "Filament-" and "Printer-"Settings, but after the last update of the system 

presets some of my "print Settings" disappearted from the list. I can see them if I export the configuration bundle. So they are not deleted. And if I try the save a new configuration with a same name as one of the lost, PrusaSlicer warns me that a configuration with that name alreasy exists. 

The "System presets" section on ALL tab is empty, too!
I have attached my configuration bundle file. 

[My config_bundle.ini file](https://github.com/prusa3d/PrusaSlicer/files/4719082/PrusaSlicer_config_bundle.zip)

rtyr, 2020-06-03:
Seems like expected behavior. In the PrusaSlicer 2.2.0 series we made a breaking change in handling compatibility between vendor profiles, vendor derived user profiles and user profiles (derived from the "-- default --" profile).

That is the reason why system profiles are not visible (they are from different vendor space, since your profiles are not inherited from system profiles).

As for the "HEVO - PETG - DasFilament - EFP 0.40" profile. This profile has compatible printer condition set to "compatible_printers = "HEVO - PETG - Bowden - EFP", which means that this profile will be visible only when "HEVO - PETG - Bowden - EFP" printer profile is selected (this printer profile is not present in provided bundle).

See release notes for PrusaSlicer 2.2.0-rc for further details. https://github.com/prusa3d/PrusaSlicer/releases. There are also some opened issues related to this, so you can continue with the discussion there. Closing this one.

closed 2020-06-03 by rtyr

ikarisan, 2020-06-03:
The "compatible printer condition" is okay. But there should be some kind of warning message if you delete a profile that has dependencies! I would expect a question like "The profile you are trying to delete has dependencies to [X], [Y]! Do you really want to remove it and dissolve its dependencies?"