prusa3d/PrusaSlicer #4341: [feature request] Set maximum temperature in machine limits

armaginus, 2020-06-03:
I operate several 3D printers. On a couple of occasions I have set the temperature too high for a couple of my printers. There really should be a way to set the maximum temperature of the machine's nozzle and/or bed in the machine limits. This would be an added minor safety measure as well in some instances and prevent some printers from becoming damaged.

BeatSlayer, 2020-06-03:
That's firmware work and most (all?) have that implemented.
And if [this page](https://reprap.org/wiki/G-code#M143:_Maximum_heater_temperature)  is correct it seems only reprap firmware is able to be re-set by Gcode.

JohnnyDeer, 2020-06-04:
Hi @armaginus , temperature limits have to be set in firmware, as @BeatSlayer wrote. Firmware also should have safety features, in this case MAXTEMP, which will save printer from being damaged. I´m closing this issue as not related to slicer.

closed 2020-06-04 by JohnnyDeer