prusa3d/PrusaSlicer #4342: Hard crash when manually changing bed size

bdbell, 2020-06-03:
### Version
2.2.0+

### Operating system type + version
macOS 10.15.4

### 3D printer brand / version + firmware version (if known)
Longer LK4
Wanhao Duplicator 6
(problem is printer independent)

### Behavior
If the bed size is changed by hand and the objects are outside of the new bed size print area, it causes a hard segmentation fault.

1. Create project based around printer with larger bed size.  In my case, the Longer LK4 has a bed size of 220x220.
2. Place an STL toward the top right of the bed area.
3. Change the bed size to a smaller bed.  In my case, the Wanhao Duplicator 6 has a bed size of 200x200.
4. Slicer will crash.

I would expect that the bed would resize and it would behave the same as if the STL is outside of the print boundaries.  This works properly if the printer is preconfigured and the bed size modifier is not altered manually.

Crash report:
Process:               PrusaSlicer [40539]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.2.0+-202003211132 (PrusaSlicer-2.2.0+-202003211132)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [40539]
User ID:               501

Date/Time:             2020-06-03 10:13:31.612 -0600
OS Version:            Mac OS X 10.15.4 (19E287)
Report Version:        12
Bridge OS Version:     4.4 (17P4281)
Anonymous UUID:        39230FC5-9B01-D9C0-161B-9583D813D9A0

Sleep/Wake UUID:       0C85AA08-3D97-417E-B352-13688BB11E2C

Time Awake Since Boot: 2500000 seconds
Time Since Wake:       7500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [40539]

JohnnyDeer, 2020-06-04:
Hi @bdbell , thank you for reporting. This issue should be solved in next version of PrusaSlicer.

FidelCapo, 2021-05-03:
This issue should be fixed in 2.3.0 and 2.3.1.
Closing

closed 2021-05-03 by FidelCapo