prusa3d/PrusaSlicer #4343: Some "defaults" may be incorrect and cause serious problems

systronix, 2020-06-03:
### Version
2.2.0+ linux-x64

### Operating system type + version
Ubuntu Linux 18.04 LTS

### 3D printer brand / version + firmware version (if known)
i3 MK2 latest FW 3.2.3

### Behavior
In slicer > Printer Settings > Advanced  > "Use relative E distances" this must be in the non-default setting (checked) state or it can cause serious print failures (apparently around retractions).

So if you are troubleshooting printer issues and try to "do the right thing" by setting everything to defaults you can shoot yourself in the foot as I did recently.

There is just the one "default" I know of which can cause this but now I am worried there may be others waiting to bite me.

Thanks for the quick reply to help me fix this on separate issue of my printer emitting a large blob after retraction.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[BearUpgrade_Y-rod-Holders_i3MK2_FM_PETG.zip](https://github.com/prusa3d/PrusaSlicer/files/4725120/BearUpgrade_Y-rod-Holders_i3MK2_FM_PETG.zip)

rtyr, 2020-06-03:
I believe you misunderstood my answer there. Default values are ok. The issue was the combination of your setting with your custom gcode (which was taken from our printer profile). So no problem on the PrusaSlicer side. I added some more explanation to #4327  as well. Closing.

closed 2020-06-03 by rtyr

systronix, 2020-06-03:
OK thanks, the tool tip is perhaps a bit misleading. See my screen capture and explanation in #4327. English is not the perfect language and precise technical communication is hard.

rtyr, 2020-06-03:
I agree, it is hard for me sometimes :-).