prusa3d/PrusaSlicer #4347: [2.2.0] Out of hand Travel, is it me? (closed a little prematurely)

FlailAway, 2020-06-04:
### Version
2.2.0

I object to what you call the travel speed "to something a it little more sensible." The PrusaSlicer produces conflicting speed changes too close together on some rare occasions. See my earlier post in this (Z-speed) issue that seemed largely written off and/or ignored.

On my delta at 100mm/s it can cause a tooth jump on the belts forcing a Layer shift, no matter how tight I have the belts. As a result of having long jobs trashed near the end with a Layer shift, I MUST use a slow travel speed. This layer occasional shift issue is not just the Delta, it does it on my CoreXY. Repeatable on both and always at the same spot in the gcode.

It only happens on long prints so now I cannot take the risk and thus, slow-travel for everything it is. Clearly the 6.5 retraction I have set is NOT being used in those specific travels instances on this print in the pic. So, it'd be nice for you to consider that I am describing an actual problem rather than just dismissing me by closing the thread.

My car only leaks when it rains, so I assume that it would be "a little more sensible to not to drive in the rain." :)

FlailAway, 2020-06-04:
Just went to the next project and the Travel is a little extreme even before I set a sensible speed. :)

[image]

rtyr, 2020-06-04:
Related to https://github.com/prusa3d/PrusaSlicer/issues/4344

1) Let me remind you your description of the previous issue. :-)
[image]

2) You can reopen closed issue and add more information.
3) z-speed. You should start by fixing the issue with your printer. I find it hard to believe, that you consider behavior of your printer normal. I don't think any further conversation about this will be fruitfull. I tried.

> Clearly the 6.5 retraction I have set is NOT being used in those specific travels instances on this  print in the pic. So, it'd be nice for you to consider that I am describing an actual problem rather than just dismissing me by closing the thread.

To the problem. As I mentioned in the previous issue, I sliced the gcode from your 3MF project, with the same version of PrusaSlicer. So I believe the resulting gcode was same as yours (you did not include it). I checked the gcode and I didn't find any missing retractions like are you suggesting. I even printed it on very similar printer, just to be sure.

It is possible, that I missed something **so, if you are familiar with the gcode structure, please show us, where are retractions missing in your opinion.** It is of course possible, that retractions actually didn't happen as a result of some failure on the printer side.

rtyr, 2020-06-05:
BTW. the issue on your photo doesn't even match travel path or any other path included in the gcode (it is the same if you preview exported gcode).

[image]

FlailAway, 2020-06-07:
Sorry, my mistake. The printer is on another computer out in the garage and no Network. I started writing the original post and saw you wanted the .3mf so had to go back out and load it up and slice again. Must have used a different profile.

There is nothing wrong with the printer as the last time I broached this layer-jumping issue I provided the exact lines of GCode that caused the problem. By manually changing the travel in that one offending line the problem went away.

Were you printing at 100mm/s?

I will go out and get another 3mf that shows the actual print above. Back soon.

FlailAway, 2020-06-07:
OK, back again and this is what I show for this attached 3mf.

[image]

[riser-02.zip](https://github.com/prusa3d/PrusaSlicer/files/4742551/riser-02.zip)

rtyr, 2020-06-07:
>  I provided the exact lines of GCode that caused the problem. By manually changing the travel in that one offending line the problem went away.

I know. But it doesn't mean that issue is in the gcode, it just means, that your printer cannot handle some situations properly (standard situations). PrusaSlicer (any slicer) just expects printer which can reliably operate within the limits set in its firmware. You are trying to treat symptoms of the problem, not the problem itself. 

BTW I am using 200mm/s travel speed on my custom Kossel without any issues. I had several configurations on this printer over the years (ramps with A4988, MKS GEN, Smoothieboard@DRV8825 and now SKR1.3+TMC2208).

rtyr, 2020-06-07:
> 
> 
> OK, back again and this is what I show for this attached 3mf.

Again, travel moves doesn't match to your photo. Similar looking travels are starting at about 26mm. Please attach the gcode too.

rtyr, 2020-06-07:
Probably not related, but I found this in my "archive". Those lines were caused by some weird bug in our firmware (also no travel moves in that area). 

[image]

rtyr, 2020-06-08:
I believe we can close this. Output from the PrusaSlicer is OK, test print is OK, strings on the photo doesn't match to any path in the gcode.

closed 2020-06-14 by rtyr