prusa3d/PrusaSlicer #4348: 180mm model detected as bigger than the surface if mirrored or rotated from original import

Thmsdc, 2020-06-04:
### Version
Prusa Slic3r 2.2.0+
202003211132

### Operating system type + version
macOS 10.15.5

### 3D printer brand / version + firmware version (if known)
Prusa Mini 4.0.5
### Behavior
When loading a .STL of 180mm long it only works in one direction. If I turn it at 90, 180, 270 degrees Prusa Slic3r says that the model is too big and getting out of the area. This block the option to slice. This issue is also occuring while mirroring.
*  Import a 180mm long model
* Rotate it through rotation tool or coordinates
* Slic3r is displaying "An object is outside the print area was detected. Resolve the current problem to continue slicing".

I guess since the bed of the Mini is supposed to accept 180 in width and height a 180mm part should fit in any direction in Slic3r.

Model is unable to be sliced in any other direction that the one after import.

[image]
[image]
[image]
[image]

JohnnyDeer, 2020-06-08:
Hi @Thmsdc , please provide 3mf file.

Thmsdc, 2020-06-08:
Please find my .3MF in attachment.

### Behavior

Made a random 180mm rectangle from a CAD software. I imported it in PrusaSlicer and same issue. At first the model fit but if I rotate it or mirror it it is not fitting anymore even back at the original position (360 degrees).

[PrusaSlicer#4348.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4744012/PrusaSlicer.4348.3mf.zip)

JohnnyDeer, 2020-06-09:
Issue reproduced and confirmed, thank you for reporting @Thmsdc .

closed 2026-09-29 by prusaslicer-bot[bot]