prusa3d/PrusaSlicer #4349: modifier for surface recycling code

kenthinson, 2020-06-04:
### Version
2.2.0+

### Operating system type + version
All Versions

### 3D printer brand / version + firmware version (if known)
All Printers

### feature request
In PrusaSlicer we can currently add modifiers. It would be great to have modifiers that would cut into the face of the part the recycle code to match our selected material type. Example I know a lot of recycle plants don't recycle PLA but almost all recycle ABS, PET, ect. 
Being able to place the imprint like we can currently move the modifier boxes would be awesome. 
I think this should be part of the slicer not the part geometry as each person will be printing with different materials.

[image]

_Is this a new feature request?_
Yes

lukasmatena, 2020-06-05:
This should be already possible, just model the icon as stl, load it as a modifier to your model, sink it into it and set it to zero perimeters, zero infill.

To illustrate this, I just embedded Prusa beer opener into treefrog's back. It may look bizarre but you get the idea.
[image]

kenthinson, 2020-06-06:
@lukasmatena Thanks for the information. I will use that for now. But still think this could be a great feature to have added instead of having to use this workaround.

Ofenhed, 2020-09-15:
I love this idea. It could also be a field in the material settings where recycling information is set, so that changing material automatically changes the recycling imprints so that you don't accidentally print with wrong recycling information as this can ruin a lot of plastic from being properly recycled.

thomasa88, 2020-09-30:
A  nifty extension would be if the model contained information on where to place the mark. (Hack: where you find the 0.0001 mm cube hole on the STL, put the mark.)

zapdogster, 2020-10-24:
I posted a similar idea in the forums and I was directed to here. I would fined it useful to have user positioned simple text embedded on the top layer. I had written:
_I would find it useful to have a very simple text embosser for marking revision numbers in the top of prints. Sometimes the tweaks to make things are so small that it is very hard to tell the difference between different test parts.

I would envision:
    - text string (X-Y positioned by user)
    - font size
    - emboss/engreave depth (Z), from the top layer
    - maybe even font type (choice of only a few?)_

**Another user posted:** It certainly would be nice to include the name of the GCode file. (especially as PS allows you to create a GCode file name that includes material and print settings etc.)

Thanks,
George

bubnikv, 2021-06-03:
In PrusaSlicer 2.4.0-alpha1 you will be able to apply a "subtractor" aka "negative volume" instead of having to use a modifier and set its top/bottom layers and infill to zero.

As a low hanging fruit, we are considering to add a gallery of basic shapes to be used for the modifiers.

kenthinson, 2021-06-03:
@bubnikv Great news! Possible the recycling designs can be included?

bubnikv, 2021-06-03:
> Possible the recycling designs can be included?

Let's see what our content guys will come with.

kenthinson, 2021-06-03:
@bubnikv It appears so.

ikjl, 2021-06-28:
I mostly print biodegradable. But I printed some petg, so I found this thread searching for ways to add recycling marks. 
It sounds like the zero fill modifier would work as a stamp at the bottom of most prints.

But it would be an awesome option to have, especially if you could set the area of a model that you want stamped and then have it change automatically depending on filament type.

bubnikv, 2021-07-30:
We have a Shape Gallery available in the upcoming PrusaSlicer 2.4.0-alpha1.

[image]

closed 2021-07-30 by bubnikv

vintagepc, 2021-09-01:
It's worth noting that PETG is not compatible with resin code 1 (PET) . The glycol modification makes it gummier than standard PET and this is problematic for some processes. 

Check your local requirements before branding your parts and chucking them in the regular recycling stream - since doing this carelessly can contaminate a batch of recycled plastic and cause more waste than if you'd just thrown it out.