prusa3d/PrusaSlicer #4354: Prusaslicer 2.2 brim issue

tazlewouaf, 2020-06-05:
Hi,
I notice a difference between version 2.1 and 2.2 of Prusa Slicer.
When I add a brim to my object, the skirt stays at the same distance from the object whereas on version 2.1 it was automatically moved outside the brim. 
Is this normal? 
Could this be taken into consideration and fixed?
[image]

Thanks

rtyr, 2020-06-05:
Yes, it is normal and intended behavior since 2.2.0.

Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3692, https://github.com/prusa3d/PrusaSlicer/issues/3923, https://github.com/prusa3d/PrusaSlicer/issues/3516, https://github.com/prusa3d/PrusaSlicer/issues/3634, https://github.com/prusa3d/PrusaSlicer/issues/4017, https://github.com/prusa3d/PrusaSlicer/issues/4024, https://github.com/prusa3d/PrusaSlicer/issues/4056, https://github.com/prusa3d/PrusaSlicer/issues/4185

closed 2020-06-05 by rtyr

tazlewouaf, 2020-06-05:
sorry for the duplicate entry and thank you for your response.

rtyr, 2020-06-05:
No problem. :-)