prusa3d/PrusaSlicer #4355: Overlapping inner perimeter walls when scaling the model

davidshin172, 2020-06-05:
### Version
2.2.0
### Operating system type + version
Arch Linux 

### 3D printer brand / version + firmware version (if known)
Ender 3 + SKR Mini E3 v1.2 + Marlin 2.0

### Behavior
* There seems to be a certain scaling percentage where inner permiter walls end up overlapping:
[image]
In the above image, there are two inner perimeters that overlap (As signified the "ridge" in the middle). This causes the wall area to be thicker than intended, leading to a bulge like so:
[image]

#### Steps to Reproduce
1. Open attached project file.
1. Note that the object is scaled at 75%.
1. Slice the object.
1. Note that the inner perimeters on the top layers are overlapping.
1. Turn scale down to 67%.
1. Note that the area previously covered as inner perimeters are now gap fills

#### Project File (.3MF) where problem occurs
[overlap.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/4737507/overlap.tar.gz)

davidshin172, 2020-06-05:
Here's the 67% one: Note the lack of bulge in the wall area (Bulge at the bottom is something I'm dealing with for this printer).
[image]

ssill2, 2021-11-24:
The bottom bulge on this is not particular to the any printer as I understand it.  The issue is documented heavily here:

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/buldge-when-print-reaches-solid-layers/paged/1/

I'm really hoping this gets some attention.

mikeev261, 2023-02-16:
Still seeing this happen in PrusaSlicer 2.5.x and now 2.6.x. 

Usually occurs with a layer change such as when infill stops, or there is a gap held up by supports (as others have noted):

Here's an example: 
[image]

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]