prusa3d/PrusaSlicer #4360: [Feature Request] Custom Extrusion Multiplier for the Topmost Layer

gihad, 2020-06-06:
### This is a new Feature Proposal

All too often I see top most surfaces that are not smooth. One common cause is when the top surface is not a shape that can be filled continuously and the slicer software optimizes the path in a way where it has to come back to meet an area that has already been filled before causing 2 lines to meet which creates a defect. If you have perfect calibration this issue is less apparent but still there. A "fix" for this is to reduce extrusion multiplier but that affects the entire print.

I'm using Cura's preview to illustrate what I mean:

[image]
[image]

The areas I circled with blue would be some areas of the print that would have this "defect".
Here are images of a printed objects showing this issue:

[image]

The makes above have the issue a bit exaggerated for illustration purposes. Lowering the extrusion multiplier in that case would make it look better.

The proposal is to allow the extrusion multiplier to be changed just for the top most layer.

A possible alternative (I can't test if it would be a viable) would be to allow the top most layer to be filled "without optimization", so the nozzle would travel the same path that it would if it was a simple surface with no circles, but it would "lift" to skip the circles.

I've uploaded the model I'm using [here](https://gofile.io/d/MUkKRW)

gihad, 2020-06-18:
nice, I can see from the live stream on youtube and the tweet that you guys took my "Alternative" suggestion from this issue. Thanks for adding this in.

- https://twitter.com/filamentfrenzy/status/1273567066654679040?s=21
- https://youtu.be/NAW2JR-D_QU?t=2293

VanessaE, 2020-06-29:
Besides the "monotonous" infill, also consider enabling playing around with ironing.

That said, there is a use for this feature request:  sometimes you HAVE to turn the flow rate down for high-precision objects, just to make sure the top is super smooth (in the absence of ironing).  So, it may be useful to reduce the flow on just the top.

gihad, 2020-07-02:
In my tests ironing doesn't remove the "lines" like monotonous infill pattern does. The feature that really helped the issue I described was monotonous infill.

I agree that there still are some use cases to be able to control the flow rate just for the topmost layer.

bubnikv, 2020-10-22:
I believe the monotonic infill solves the issue.
Closing.

closed 2020-10-22 by bubnikv