prusa3d/PrusaSlicer #4361: Wipe tower extends beyond platform

readmodifywrite, 2020-06-06:
2.2.0

### Operating system type + version
Linux Mint 19.3

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S
FW 3.9.0

### Behavior

Wipe tower loses its mind:

[image]

rtyr, 2020-06-06:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4192 and https://github.com/prusa3d/PrusaSlicer/issues/3919.

It is a known bug in the Prusaslicer 2.2.0. It will be fixed in the next PrusaSlicer release. In the meantime, there is a simple workaround to prevent this. Set your max layer height limit (printer settings) to a slightly higher value than regular layerheight in your print profile. I guess you are using 0.3 DRAFT, so you can set layer height limit to 0.31.

Closing.

[image]

closed 2020-06-06 by rtyr

readmodifywrite, 2020-06-07:
Thank you so much!  And sorry for the dupe!